libremesh/lime-packages #5: Ebtables/IPtables without firewall application

p4u, 2015-01-14:
If the openwrt firewall application is not installed, ebtable rules are not properly executed. An autonomous system application not firewall dependent is needed to control ebtables/iptables.

ilario, 2015-09-01:
Considering the use of *lime-proto-wan* in absence of OpenWRT firewall application, there is still need to manually write rules for masquerading, right? 
Should this re-open this issue?

ilario, 2015-09-02:
What I mean is something like [this](https://github.com/libre-mesh/lime-packages/blob/0a29a26894ffb73366bc6d6d2b1e8a3d85e54b24/packages/bmx6-auto-gw-mode/files/etc/watchping/wan-ok.d/bmx6-gw#L10).

altergui, 2015-09-09:
p4u, dear, 
loving your coding as always, and every commit signed by you is greatly appreciated :)

but could we please not have (again!) a "lime-init" thing? :P

i was happy i killed it with
https://github.com/libre-mesh/lime-packages/commit/854c93744e

and even offered an explanation on
https://github.com/libre-mesh/lime-packages/commit/f65c4d6ea7

"[lime-init is] unused and unnecesarily duplicates /etc/init.d/ functionality
if in doubt, check for example /etc/init.d/done that's really minimal (and actually is in charge of running /etc/rc.local as a monolithic hook). Anything that would go in /etc/rc.local.d/* can be implemented with just a few more lines as an /etc/init.d script, keeping LiMe closer to upstream architecture."

so i'd vote to turn this (reincarnation of) lime-init into a specific /etc/init.d/firewall-user, that only runs "sh /etc/firewall.user" at start.

that's what /etc/init.d/ is there for, after all ;)

if no-one else steps up with this, i can do it on my next coding iteration

ilario, 2015-10-12:
Referring to my previous comment, I included some code from bmx6-auto-gw-mode in batman-adv-auto-gw-mode, [here](https://github.com/libre-mesh/lime-packages/pull/20) you are the pull request.

p4u, 2016-02-11:
Are we in a loop? How is this issue now?

trizonelabs, 2016-03-15:
ebtables : please could someone give me a hint why this was introduced and where this is used ?

p4u, 2016-03-15:
Check this http://wiki.libre-mesh.org/doku.php/net_architecture

An look for "ebtables".

p4u, 2016-07-21:
So, if firewall is not enabled ebtables rules are not applied. This ends up in a very strange behavior. 

So commit  dcc5d5ef44dcf11e0f9e42808fa195164a91d00e fixes this. If there is a better option than including a init script, then I would like to know.

closed 2016-07-21 by p4u