libremesh/lime-packages #6: BMX6 do not configure properly ethernet devices

p4u, 2015-01-16:
While the name of the VLAN devices is now eth0-13 in the network configuration, bmx6 still uses the old format eth0_13. It seems to affect only ethernet interfaces.

p4u, 2015-01-16:
ip link show| grep eth

2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel master br-lan state UP mode DEFAULT group default qlen 1000

3: eth1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel master br-lan state DOWN mode DEFAULT group default qlen 1000

10: eth0-126@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1496 qdisc noqueue master bat0 state UP mode DEFAULT group default 

11: eth0-13@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1398 qdisc noqueue state UP mode DEFAULT group default 

12: eth1_126@eth1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1496 qdisc noqueue state LOWERLAYERDOWN mode DEFAULT group default 

13: eth1_13@eth1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1398 qdisc noqueue state LOWERLAYERDOWN mode DEFAULT group default

p4u, 2015-01-16:
uci show network | grep ifname | grep eth

network.lan.ifname=bat0 eth0 eth1
network.lm_eth0_batadv_dev.ifname=eth0
network.lm_eth0_batadv_if.ifname=eth0-126
network.lm_eth0_bmx6_dev.ifname=eth0
network.lm_eth0_bmx6_if.ifname=eth0-13
network.lm_net_eth0_batadv_dev.ifname=eth0
network.lm_net_eth0_batadv_if.ifname=eth0_126
network.lm_net_eth0_bmx6_dev.ifname=eth0
network.lm_net_eth0_bmx6_if.ifname=eth0_13
network.lm_net_eth1_batadv_dev.ifname=eth1
network.lm_net_eth1_batadv_if.ifname=eth1_126
network.lm_net_eth1_bmx6_dev.ifname=eth1
network.lm_net_eth1_bmx6_if.ifname=eth1_13

p4u, 2015-01-16:
uci show bmx6 | grep eth

bmx6.lm_net_eth0_bmx6_if=dev
bmx6.lm_net_eth0_bmx6_if.dev=eth0_13
bmx6.lm_net_eth1_bmx6_if=dev
bmx6.lm_net_eth1_bmx6_if.dev=eth1_13

p4u, 2015-01-16:
As it can be seen eth0_13 does not exist (but eth0-13 yes)  so bmx6 does not use this interface properly.

G10h4ck, 2015-08-31:
newer images seems un'affected, i close this for the moment

closed 2015-08-31 by G10h4ck