libremesh/lime-packages #7: luci.ip.Hex() does not exist anymore

p4u, 2015-01-29:
luci.ip.Hex() function has been removed from the luci libraries.

https://github.com/libre-mesh/lime-packages/blob/develop/packages/lime-system/files/usr/lib/lua/lime/network.lua#L37

IRC log about this issue:

14:02 < p4u> lua from Jan 15: ip.lua -> function Hex( hex, prefix, family, swap )
14:03 < p4u> now I see all the libraries have been moved and some of then even rewritten
14:03 < p4u> the ip module is missing the Hex() function
14:03 < p4u> as descreibes the documentation
14:03 < p4u> jow_laptop: what you mean for input data?
14:03 -!- iamfrankenstein [~Thunderbi@h90192.upc-h.chello.nl] has quit [Ping timeout: 245 seconds]
14:03 < jow_laptop> p4u: yeah I dropped Hex() because I thought only luci uses it internally for parsing /proc/net/ipv6_route
14:04 < jow_laptop> p4u: therfore I wanted to know what kind of data you feed to hex, to see if you can use another variant or if I should reimplement the Hex() constructor
14:04 < p4u> https://github.com/libre-mesh/lime-packages/blob/develop/packages/lime-system/files/usr/lib/lua/lime/network.lua#L37
14:05 < p4u> we were using it ;(
14:05 -!- BankZ- [~bankz@c-73-161-152-148.hsd1.mi.comcast.net] has quit [Ping timeout: 252 seconds]
14:08 < jow_laptop> hm, ok I can readd Hex() but a few other parts of the code will break too because you access private internals of the cidr object
14:08 < p4u> ok
14:09 < p4u> we can fix it
14:09 < p4u> or maybe find another way for doing the same

G10h4ck, 2015-01-29:
Is this bug present in last stable or just in trunk?
If it is just in trunk i would wait that trunk stabilize a little before 
changing stuff in libre-mesh to fix stuff like this

P.S. please keep dev@lists.libre-mesh.org in CC

p4u, 2015-01-29:
It is not trunk neither stable, it is just "LUCI", the repository
"http://git.openwrt.org/project/luci.git". We are using it in all LiMe
Branches.

p4u, 2015-02-01:
On Thursday, January 29, 2015 02:34:59 PM Pau wrote:
> It is not trunk neither stable, it is just "LUCI", the repository
> "http://git.openwrt.org/project/luci.git". We are using it in all LiMe
> Branches.

No it is trunk, stable is using luci stable version src-git luci 
https://github.com/openwrt/luci.git;luci-0.12

note the ending ;luci-0.12 so it is not a bug for the lime stable version, 
it's trunk specific ;)

P.S. I remember we agreed to use http://libre-mesh.org/ to manage the project, 
so we should keep bug there

ilario, 2015-10-24:
This issue is critical for switching to nowadays OpenWRT stable. 
I think we should change our code in network.lua rather than include the whole luci.ip file as proposed in branch ip_legacy_module.

ilario, 2015-10-24:
From [that](https://github.com/openwrt/luci/blob/luci-0.12/modules/base/luasrc/ip.lua) module we were using also functions IPv4() and IPv6(). Writing ourself an equivalent code seems quite time expensive... I will make a package which includes just this library.

ilario, 2015-10-24:
I [moved](https://github.com/ilario/lime-packages/commit/d506692467037338561cd55eac7dbc768c9b766d) the legacy luci.ip to a new package called libluciip-lua and added this to lime-system dependencies, you can find out the new package [here](/ilario/lime-packages/tree/ip_legacy_module/packages/libluciip-lua).

ilario, 2015-11-02:
This has been solved by @G10h4ck in the new [luci-lib-ip branch](/libre-mesh/lime-packages/tree/hotfix/luci-lib-ip which has been merged in develop branch. Please close this issue.

closed 2015-11-02 by G10h4ck