libremesh/lime-packages #17: Proto without 802.11ad VLANs

gabri94, 2015-09-16:
As describing in #16, when u try to create an olsr configuration without using any vlan (olsr:0) the network device is not created.

For example if i declare the network as
<code>
config net port5                                                           
	option linux_name 'wlan0_adhoc'                                          
	list protocols 'olsr:0'                                               
</code>

When i generate the uci config with lime-config, wlan0_adhoc is not declared in /etc/config/network

How should i use a routing protocol without any vlan?

G10h4ck, 2015-09-17:
I would expect lime.proto.adhoc to create the base openwrt interface for wlan0_adhoc, but I am not sure if something similar happen also for plain Ethernet interface...
No time for further analysis right now, merge requests are welcome

ilario, 2015-11-25:
With help from Gio, I started to address the problem [here](/ilario/lime-packages/compare/sandbox/olsr-without-vlan).
Does the olsr6 also need an IP on the interface for running?

We planned to solve this problem in a routing protocol independent way using macvlan, in the future.

G10h4ck, 2015-11-27:
I have created an helper function to create macvlan without knowing the details or doing copy paste of code, take a look at this branch that is not tested yet

https://github.com/libre-mesh/lime-packages/tree/sandbox/macvlan

gabri94, 2015-11-27:
How macvlan is going to solve this bug?
We would like to use LiMe in a legacy WCN with heterogeneous operating systems, so without any vlan on the 802.11 side.

G10h4ck, 2015-11-27:
macvlan doesn't tag or alter packets so is compatible with other OS, btw unnamed_search_motor is your friend

ilario, 2016-05-02:
looks that this issue is being addressed in branch vlanzero, check it out [here](/libre-mesh/lime-packages/commits/sandbox/vlanzero)

G10h4ck, 2016-05-04:
fixed in fae59eda7464ccfd8a0a806295084ccb11485e82

closed 2016-05-04 by G10h4ck