libremesh/lime-packages #18: Separate settings for netmask and IP range auto assignment

ilario, 2015-09-27:
I like the flexibility of the [main_ipvX_addresses auto completing feature](https://github.com/libre-mesh/lime-packages/blob/develop/packages/lime-system/files/etc/config/lime#L22) but I would **separate the netmask from the IP auto assignment range**.
Is there a way to **specify a different netmask**?
For example, [I used](https://github.com/VeronaWirelessCommunity/lime-packages/blob/develop/packages/lime-system/files/etc/config/lime-defaults#L11):

     option main_ipv4_address '10.37.188.0/22'

and it works very well for auto assigning the IP, but I did want to have a /16 netmask on this IP...
Thanks

G10h4ck, 2015-09-27:
What happen if you substitute 22 with 16 in your configuration ? What do you expected instead?

ilario, 2015-09-28:
> What happen if you substitute /22 with /16 in your configuration ?
> What do you expected instead?

If I use
    option main_ipv4_address '10.37.188.0/22'
I get
* IPv4 auto completed in range 10.37.188.1-10.37.191.254
* netmask /22

If I use
    option main_ipv4_address '10.37.0.0/16'
I get
* IPv4 auto completed in range 10.37.0.1-10.37.255.254
* netmask /16

But I do want
* IPv4 auto completed in range 10.37.188.1-10.37.191.254
* netmask /16

> Moreover I just noticed you are trying to use a /72 for ipv6 this cannot work 
> the minimum required is a /64 subnet, if you need more then one just take it from librenet6.

Same discourse as for IPv4, I want:
* IPv6 auto completed in range 2a00:1508:0001:f015:ef00:0000:0000:0000-2a00:1508:0001:f015:efff:ffff:ffff:ffff and I get this setting 2a00:1508:1:f015:ef00::/72
* netmask /64

ilario, 2015-10-12:
So, my problem is that in the /16 subnet I have also some static IPs, so the range of IP autocompletion should be smaller than the whole subnet.

I propose to add an **optional** field in *main_ipv4_address* separated by a slash, the first field is the IP or network, the second is for autocompletion and the third is for the netmask. If just two fields are present the second will be used for both concepts, as happens now. So in my case it would result:

    option main_ipv4_address '10.37.188.0/22/16'

obtaining
* IPv4 auto completed in range 10.37.188.1-10.37.191.254 (indicated by /22)
* netmask for subnet /16

If it seems ok to you I think I can implement this.
Let me know!

ilario, 2015-10-13:
I'm working to implement my proposal, still a senseless error when issuing uci commit. As soon as I finish I will do a pull request.

ilario, 2015-10-16:
Oh, the error I got was simply because I forgot to uncomment the section name in lime config file... Maybe these section names could be uncommented by default?
Anyway, the patch is working.

ilario, 2015-10-24:
This was solved in #22 and documented in [this](/libre-mesh/lime-packages/blob/bf4f8f77aa42334f4d867d6a5cdaa1817108bef4/packages/lime-system/files/etc/config/lime#L22) inline comment.
The usage mode is slightly different from the one that was suggested here. Now the **first** parameter sets the **subnet** and the **second** sets the **range for auto-completion** of the IP.

closed 2015-10-24 by ilario