libremesh/lime-packages #28: bmx6 NOT selected builds/installes  bmx6 libs

trizonelabs, 2015-11-01:
make menuconfig bmx6 deselcted builds bmx6 related blobs into the image.
As its quite big would be better to inhibit this via deselected bmx6.

3zl

ilario, 2015-11-02:
Could you rephrase this? 
I'm not used to bmx6 so I didn't get the problem.
Thanks,
Ilario

trizonelabs, 2015-11-02:
Ilario tnx for response .
configs from last build 
 
build ESXi 20151102.1707

feeds:
src-git lime https://github.com/libre-mesh/lime-packages.git;sandbox/release/15.09

diffconfig:
Routing and Redirection
CONFIG_PACKAGE_bmx6=y

CONFIG_PACKAGE_luci-app-bmx6=y
CONFIG_PACKAGE_luci-lib-libremap-bmx6=y

.config
LiMe
# CONFIG_PACKAGE_bmx6-auto-gw-mode is not set
LUCI 
	3.Applications 
CONFIG_PACKAGE_luci-app-bmx6=y
	8.Libraries
CONFIG_PACKAGE_luci-lib-libremap-bmx6=y

menuconfig search :
Symbol: PACKAGE_luci-app-bmx6 [=y]                                                                      

  │  
  │ Type  : tristate                                                                               

           │  
  │ Prompt: luci-app-bmx6........ bmx6 configuration, status and visualization module     

                    │  
  │   Location:                                                                  

                             │  
  │     -> LuCI                                                         

                                      │  
  │ (1)   -> 3. Applications                                   

                                               │  
  │   Defined at tmp/.config-package.in:29668         

                                                        │  
  │   Selects: PACKAGE_libssp [=n] && 

PACKAGE_bmx6 [=y] && PACKAGE_luci-lib-json [=y] && PACKAGE_luci-mod-adm │  
  │   Selected by: 

PACKAGE_lime-full [=n] && IPV6 [=y] && !PACKAGE_dnsmasq [=y] || PACKAGE_lime-webui [=y] 

so its seems bmx6 is selected though lime_webui ?
I could not find another place where it is activated.

(sorry for the format , but editing text quite combersome)

cheers 
3zl (Reinhard)

ilario, 2015-11-02:
Hey you closed the issue, I suppose by error.

Anyway you're right, selecting lime-webui you get luci-app-bmx6 selected, which selects bmx6. This is super annoying also for me. And the same problem happens with batman-adv when you're not using it.

I think this can be easily solved inserting [conditional dependencies](http://wiki.openwrt.org/doc/devel/dependencies#dependency_types) in [lime-webui Makefile](/libre-mesh/lime-packages/blob/develop/packages/lime-webui/Makefile#L24), for example substituting:

    DEPENDS:= luci-app-bmx6

with:

    DEPENDS:= +PACKAGE_lime-proto-bmx6:luci-app-bmx6

G10h4ck, 2015-11-02:
Pull requests are welcome

trizonelabs, 2015-11-02:
sorry for the fault - will test

cheers 3zl (Reinhard)

ilario, 2016-05-02:
this should have been solved by the merge of #40

G10h4ck, 2016-05-04:
Addressed in 234191a90a5fb7173e2756c851ee40d5b75c90ec and in 234191a90a5fb7173e2756c851ee40d5b75c90ec

ilario, 2016-08-23:
This issue resurrected with 668c41f by @p4u which more or less reverted 234191a.

p4u, 2016-09-01:
luci-app-bmx6 was not installed in generic images created with collection lime-full, it was a problem, so I added it back to lime-webui. However it understand it may be annoying for someone who wants to use lime-webui without bmx6. I did not know about the conditional dependencies, so it seems like a good work around for this issue (add to lime-webui Makefile +PACKAGE_lime-proto-bmx6:luci-app-bmx6)

Can you confirm it works fine?

G10h4ck, 2016-09-02:
yeah it worked fine

p4u, 2016-09-03:
Fixed on 888d6cd5c96547c26407e00e729a004833e3e8e2

closed 2016-09-03 by p4u