libremesh/lime-packages #29: libremap confif

trizonelabs, 2015-11-18:
libremap-agent-openwrt/libremap-agent/files/etc/uci-defaults/ is setting a /etc/config/libremap file if there is none existing, alas the config file is generated by libre-map though /files/config/libremap with different uci parameters.

is there a reason for libre-mesh having his own config/libremap ?

please someone should have an eye on this because it seems to generates some problems setting our own parameters via scripts.

G10h4ck, 2015-12-04:
Yes it is the expected behaviour, libremap-agent is firmware agnostic so libre-mesh create a more specific configuration that fits better in LiMe

ilario, 2016-08-31:
So? Is this still an issue?

closed 2016-09-02 by G10h4ck