libremesh/lime-packages #30: libremap-agent plugin error in LUCI master

trizonelabs, 2015-12-02:
running libremap-agent results in log error messages 
1. plugin system sysinfo nil
2. plugin wireless  iwdata nil

i did have a look at the source but with nov docs about the structure im lost

does anyboday have any idea or patch
cheers
3zl

trizonelabs, 2015-12-04:
taking a closer look this is a problem libremap-agent not ported to LUCI 15x(master).
major changes have been made since LUCI 0.12 (Jan.2015).
we will fork libremap-agend and try to port it to to LUCI  resp. CC.
Any help would be welcome
cheers 
3zl

ilario, 2016-08-31:
So is this a issue of [libremap-agent](/libremap/libremap-agent-openwrt)? 
Can you file it there and close it here?

closed 2016-09-02 by G10h4ck