libremesh/lime-packages #32: Feature request: olsr-auto-gw-mode

ilario, 2015-12-05:
Just a reminder: 
we need a package equivalent to *batman-adv-auto-gw-mode* and *bmx6-auto-gw-mode* working for OLSR.

FreifunkUFO, 2016-07-15:
f.i. check the years-old freifunk gw-check: https://github.com/openwrt/luci/blob/master/contrib/package/freifunk-gwcheck/files/usr/sbin/ff_olsr_test_gw.sh

nicopace, 2017-03-31:
@ilario is this still needed? the issue has some time around and had noone to take care of it...

Could it be that there are no users that need it?

ilario, 2017-04-23:
Yep, that's the case, up to now we have no user (AFAIK) who is using OLSR or OLSR2.
That's a pity 'cause it's implemented, except for auto-gateway.
Back then we supposed people from Ninux in Italy would have needed this.
@gabri94 any opinion of this?

gmarcos87, 2019-03-06:
None of the stakeholders took the initiative. I close this until there is a real demand for this feature. The issue was opened more than three years ago.

closed 2019-03-06 by gmarcos87