libremesh/lime-packages #33: Feature request: Specify DHCP range

ilario, 2015-12-24:
Hi!
I want to compile a Libre-Mesh firmware image which contains a custom IP range (= 10.37.192.1 - 10.37.255.254) for the DHCP server.
To specify this in the feeds/lime/packages/lime-system/files/etc/config/lime-defaults would be optimal.
Up to now the workaround is to modify the start (= 49152) and limit (= 16382) options in packages/network/services/dnsmasq/files/dhcp.conf which will end up to be /etc/config/dhcp.
Thanks,
Ilario

G10h4ck, 2016-07-05:
Wasn't this solved by your pull request some time ago?

ilario, 2016-07-06:
No, absolutely. The problem is still there.

ilario, 2016-08-08:
In LiMeCat2016q3 we discussed this and decided that `/etc/config/lime*` files are just for "one to many" settings (with a few exceptions like hostname), so in this case writing directly in /etc/config/dhcp is not harder than writing in /etc/config/lime.

ilario, 2017-03-03:
We discovered that writing to `/etc/config/dhcp` is now enough as the DHCP range has been defined in 3a6596d as the full broadcast domain in anygw package.

nicopace, 2017-03-31:
@ilario you said tha was enougth, but you reopened the issue!
What was missing?
There is also a related issue #106

ilario, 2017-04-23:
@nicopace the DHCP range is still hardcoded in anygw code (see above), I didn't dare to try coding something for changing this...
Now you just can't change the DHCP range, that's why I reopened the issue :)

nicopace, 2017-04-23:
Get it ... let's see what happens with #106

ilario, 2017-04-23:
I see the thing the other way around: let's fix this and #106 will be automagically fixed also.

nicopace, 2017-04-23:
Yeah... you are right @ilario... in fact, it seems this one and #106 are duplicates, right?

ilario, 2017-04-23:
In my opinion yes :)

ilario, 2017-08-31:
The anygw code changed but looks the issue is still there (to be verified that /etc/config/dhcp is still ignored), specifically the value is set in these two lines introduced in f40accd:
```
uci:set("dhcp", "lm_net_br_lan_anygw_dhcp", "start", "2")
uci:set("dhcp", "lm_net_br_lan_anygw_dhcp", "limit", (2 ^ (32 - anygw_ipv4:prefix()))) -- use whole network
```

ilario, 2017-09-05:
Do you think it's ok to add a couple of options like `anygw_dhcp_start` and `anygw_dhcp_limit` to network section in `/etc/config/lime-defaults`?
An alternative is to set this with the subnet notation (like an `option anygw_dhcp '10.%N1.128.0/17'`).

ilario, 2017-09-08:
I propose an **untested** solution in #222.

ilario, 2017-09-08:
An alternative would be to encourage users to customize values in `/etc/config/dhcp`. Taking the raw configs directly from there would allow users to use other options currently unavailable with anygw, for example static DHCP leases.

closed 2017-09-15 by p4u

ilario, 2017-09-22:
Don't you think a similar option should be implemented for IPv6?

PR #222 (2017-09-15): Added option for setting DHCP pool limits, fix #33
This PR is **untested**, please test this before merging. Should fix #33.
commit 918c0ad2 (2017-09-15): Added option for setting DHCP pool limits, fix #33
commit cc00e58c (2017-09-15): Merge pull request #222 from ilario/issue/33

Added option for setting DHCP pool limits, fix #33