libremesh/lime-packages #39: lime & luci packet dependencies

trizonelabs, 2016-03-09:
we do have a hard time to find & control the dependencies which seems to be messed up.
trying to build for 4MB devices with CC we always run in overhead of bildin packets.

exmpl : * bxm , ipv6..a.s.o*

 please can someone to a grep / sort & dependency tree to rebuild the menuconfig points
& check if they are really necessary.

The openwrt ipv6 global build setting / 'Enable IPv6 support in packages' could come quite handy if implemented in the Makefile

we do have to do this manually with every new feeds - update

seems to us there is a sever lack of consolidation in project base.
sorry but we dont have the knowledge & manpower for pull requests because we have to concentrate our effords at the application layer.

The additional work makes it very difficult to keep lime into our standard.setup

regards 3zl /  i4free-gr

ilario, 2016-03-09:
> dependencies which seems to be messed up

We had a huge merge in February 28th 761e9c8, changing also some Makefile. @lime-devs could this be related?

> overhead of bildin packets. exmpl : * bxm

I suppose this, regarding BMX6, is caused by #28 and analogous problems in other Makefiles.

ilario, 2016-03-09:
> The openwrt ipv6 global build setting / 'Enable IPv6 support in packages' could come quite handy if implemented in the Makefile

For me `CONFIG_IPV6` which is "Global build settings > Enable IPv6 support in packages (passes --enable-ipv6 to configure scripts)." is selected by default...

G10h4ck, 2016-03-27:
I can't reproduce this problem, 3zl can you do a double check ?

closed 2016-05-04 by G10h4ck