libremesh/lime-packages #47: WiFi client mode

p4u, 2016-05-09:
In addition to ap, adhoc and 802.11s, it would be useful to have also the client mode.

Example of configuration for /etc/config/lime

```
config wifi radio0
	list modes 'client'
	list protocols 'wan'
	option channel_2ghz 'auto'
	option client_ssid 'WLAN1'
	option client_key '9781saf84q09asf1'
	option encryption 'psk'
```

ilario, 2016-05-09:
Sounds very useful! 
But you mention that we have an "ap" mode... Do you mean the ap interface for non-lime clients connection? On that interface there is no routing protocol running, right? 
If it's so, we would need to add also an ap mode with routing protocols.
It's good to test alternatives to ad-hoc, its implementation in current hardware and drivers could be much worse than AP/STA implementation (for example I could find absolutely **no** mention to ad-hoc mode in 802.11n certification related documentation, for example [here](http://unina.stidue.net/Protocolli%20per%20Reti%20Mobili/Materiale/resource_2/802.11n_White_Paper.pdf)).

p4u, 2016-05-09:
In one hand we have the "mode" which might be: ap, adhoc or 802.11s

In the other we have the "protocol" which might be: wan, lan, bmx6...

So we are talking about different layers

ilario, 2016-05-10:
Ok.
So, on current LiMe, configuring something like

```
config wifi radio0
	list modes 'ap'
	list protocols 'bmx6:13'
	option channel_2ghz '1'
	option ap_ssid 'libre-mesh'
```
does work? Good to know!!

Regarding 802.11s, is it already supported in LiMe?

G10h4ck, 2016-05-11:
It is easy to implement but you should do two config sections because wifi and net are two separated layers

config wifi radioX
    list modes 'sta'
    ....

config net wlan_client
    option linux_name wlanX_sta
    list protocols wan

Cheers!

p4u, 2016-09-02:
Partially added on commit 7510ba643d766986607fc349825d311df8f16e7a

ilario, 2016-10-19:
What is missing to this feature to be considered completed?

nicopace, 2017-03-31:
@p4u could you check if this is already implemented, or state what is missing?
Thanks! :)

ilario, 2017-04-23:
I and @G10h4ck think that the hardcoded WAN network for client mode should be removed, even if this requires the user to write a second configuration section.
Client mode could be used also for LAN (building backbone) and not only for WAN (reaching internet).
The line is [this one](https://github.com/libremesh/lime-packages/blob/7510ba643d766986607fc349825d311df8f16e7a/packages/lime-system/files/usr/lib/lua/lime/mode/client.lua#L5):
`args["network"] = "wan"`

ilario, 2017-05-12:
@p4u do you agree to remove the hardcoded WAN setting and to expand the documentation for explaining that the additional network setting is required?

p4u, 2017-05-12:
I think wan should be the default option. So the mentioned sentence might be something like

`if args["network"] == nil then args["network"] = "wan"`

Cheers.

ilario, 2017-05-12:
Sounds good for me!! @G10h4ck ?

G10h4ck, 2017-05-14:
It's ok conceptually, but it doesn't fit in the architecture that way, I'll think of a way to do it

Cheers!

closed 2017-05-18 by G10h4ck

ilario, 2017-07-23:
@p4u @G10h4ck how can I predict the linux_name of the client interface?
This is needed for writing an example of documentation.
For making it clear, an example of minimal client configuration could be:

```
config wifi wificlientradio
	list modes 'client'
	option channel_2ghz 'auto'
	option client_ssid 'SomeWiFiNetwork'
	option client_key 'SomeWPApskPassword'
	option client_encryption 'psk'

config net wificlientnet
	option linux_name '??????????????????'
	list protocols 'wan'
```

what to use for linux_name?

ilario, 2017-07-23:
This discussion better fits in #127, copying there.