libremesh/lime-packages #61: unregister_netdevice problems on openwrt-cc

FreifunkUFO, 2016-07-25:
when using openwrt-cc instead of openwrt/lede-trunk (so its an older kernel) we should take care about batman-adv kernel problems. (seen on tplink 710v2.1)  a daily reboot would be needed to avoid loosing meshnodes, but it shell be at least "reboot -f" instead of "reboot".

with p4us build (4 days ago) i got:

dhoc_262 to become free. Usage count = 1
[ 1221.890000unregister_ne] tdevice: waiting for wlan0-adhoc_262 to become free. Usage count = 1
[ 1232.030000] unregister_netdevice: waiting for wlan0-adhoc_262 to become free. Usage count = 1
[ 1242.170000] unregister_netdevice: waiting for wlan0-adhoc_262 to become free. Usage count = 1
[ 1252.310000] unregister_netdevice: waiting for wlan0-adhoc_262 to become free. Usage count = 1
[ 1262.450000] unregister_netdevice: waiting for wlan0-adhoc_262 to become free. Usage count = 1
[ 1272.590000] unregister_netdevice: waiting for wlan0-adhoc_262 to become free. Usage count = 1
[ 1282.730000] unregister_netdevice: waiting for wlan0-adhoc_262 to become free. Usage count = 1
[ 1292.870000] unregister_netdevice: waiting for wlan0-adhoc_262 to become free. Usage count = 1
[ 1297.100000] IPv6: ADDRCONF(NETDEV_UP): wlan0-ap: link is not ready
[ 1297.120000] device wlan0-ap entered promiscuous mode
[ 1297.160000] br-lan: port 3(wlan0-ap) entered forwarding state
[ 1297.160000] br-lan: port 3(wlan0-ap) entered forwarding state
[ 1297.170000] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0-ap: link becomes ready
[ 1297.380000] IPv6: ADDRCONF(NETDEV_UP): wlan0-adhoc: link is not ready
[ 1297.400000] wlan0-adhoc: Created IBSS using preconfigured BSSID ca:fe:00:c0:ff:ee
[ 1297.410000] wlan0-adhoc: Creating new IBSS network, BSSID ca:fe:00:c0:ff:ee
[ 1297.420000] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0-adhoc: link becomes ready
[ 1299.160000] br-lan: port 3(wlan0-ap) entered forwarding state
[ 1528.570000] ip6_tunnel: bmxmain xmit: Local address not yet configured!
[ 1528.720000] ip6_tunnel: bmxmain xmit: Local address not yet configured!
[ 1529.550000] ip6_tunnel: bmxmain xmit: Local address not yet configured!
root@LiMe-c19427:/# bmx6 -cd8
status:
version        compat name        primaryIp                        tun6Address                       tun4Address      uptime     cpu nodes 
BMX6-0.1-alpha 16     LiMe-c19427 fd66:66:66:8:a62b:b0ff:fec1:9427 fdf4:3407:f64b:2400::2794:c100/64 10.246.148.39/16 0:00:00:54 0.1 1    
interfaces:
devName        state type     rateMin rateMax llocalIp                     globalIp                            primary 
eth0_13        UP    ethernet 1000M   1000M   fe80::a62b:b0ff:fec1:9427/64 fd66:66:66:8:a62b:b0ff:fec1:9427/64 1       
wlan0-adhoc_13 DOWN  INACTIVE INVALID INVALID /-1                          /-1                                 0       
originators:
name        blocked primaryIp                        routes viaIp viaDev metric lastDesc lastRef 
LiMe-c19427 0       fd66:66:66:8:a62b:b0ff:fec1:9427 0      ::    ---    128G   52       0      
tunnels:
name     net           min max hyst rating minBw tunName tunRoute remoteName advNet advBw pathMtc tunMtc 
clouds   10.0.0.0/8    8   128 20   100    960   ---     ---      ---        ---    ---   ---     ---    
inet4    0.0.0.0/0     0   0   20   100    960   ---     ---      ---        ---    ---   ---     ---    
inet6    ::/0          0   0   20   100    960   ---     ---      ---        ---    ---   ---     ---    
nodes    172.16.0.0/12 12  128 20   100    960   ---     ---      ---        ---    ---   ---     ---

p4u, 2016-07-26:
Discussion on #50

FreifunkUFO, 2016-08-05:
please re-open this ticket..
the *unregister_netdevice* problem with openwrt-cc is NOT solved yet (even when using quick-workarounds like daily-reboot)!

FreifunkUFO, 2017-03-13:
problem might be gone with newer openwrt/lede :-/

closed 2017-03-13 by FreifunkUFO