libremesh/lime-packages #62: Recursive dependencies

p4u, 2016-07-25:
Some recursive dependencies of lime-packages might be fixed.

```
make[1]: Entering directory `/home/lime/lime-build-lede/build/src'
tmp/.config-package.in:31200:error: recursive dependency detected!
tmp/.config-package.in:31200:   symbol PACKAGE_lime-proto-batadv is selected by PACKAGE_lime-system
tmp/.config-package.in:31387:   symbol PACKAGE_lime-system is selected by PACKAGE_lime-proto-batadv
tmp/.config-package.in:31235:error: recursive dependency detected!
tmp/.config-package.in:31235:   symbol PACKAGE_lime-proto-bmx6 is selected by PACKAGE_bmx6-auto-gw-mode
tmp/.config-package.in:30937:   symbol PACKAGE_bmx6-auto-gw-mode depends on PACKAGE_bmx6
tmp/.config-package.in:55558:   symbol PACKAGE_bmx6 is selected by PACKAGE_lime-proto-bmx6
tmp/.config-package.in:31258:error: recursive dependency detected!
tmp/.config-package.in:31258:   symbol PACKAGE_lime-proto-bmx7 is selected by PACKAGE_bmx7-auto-gw-mode
tmp/.config-package.in:30955:   symbol PACKAGE_bmx7-auto-gw-mode depends on PACKAGE_bmx7
tmp/.config-package.in:55633:   symbol PACKAGE_bmx7 is selected by PACKAGE_lime-proto-bmx7
warning: (PACKAGE_lime-basic) selects PACKAGE_bmx6-auto-gw-mode which has unmet direct dependencies (PACKAGE_bmx6)
warning: (PACKAGE_lime-proto-bmx6) selects PACKAGE_bmx6-uci-config which has unmet direct dependencies (PACKAGE_bmx6)
warning: (PACKAGE_lime-proto-bmx6) selects PACKAGE_bmx6-json which has unmet direct dependencies (PACKAGE_bmx6)
warning: (PACKAGE_lime-proto-bmx6) selects PACKAGE_bmx6-sms which has unmet direct dependencies (PACKAGE_bmx6)

```

ilario, 2016-07-25:
This behavior has been discussed in #41 and reported as a bug in OpenWrt Trac [here](https://dev.openwrt.org/ticket/22226).

p4u, 2016-07-25:
Right, did not see the other thread.

closed 2016-07-25 by p4u