libremesh/lime-packages #66: Use pingcheck instead of watchping

p4u, 2016-08-06:
A new OpenWRT/LEDE package written in C has been released https://github.com/br101/pingcheck

It uses UBUS to communicate the current status and is compatible with UCI.

It seems a much better option than the current watchping used in LiMe for bmx6 auto gw mode and some other packages.

FreifunkUFO, 2017-03-13:
pingchecks will have problems with captive-portal-splashed internet-gateways. So ping may see "internet", while access is still forbidden/splashed.

so for olsr-freifunk there is a very old package, which tries to get a http-website, also with treshold
https://github.com/openwrt/luci/tree/master/contrib/package/freifunk-gwcheck

aparcar, 2018-01-23:
If pingcheck does [not support ipv6](https://github.com/br101/pingcheck/issues/1) and a http crawl as requested by @FreifunkUFO we may should simply extend watchping and close this issue?

G10h4ck, 2018-01-23:
It seems a good starting point for a LiMe related GSoC proposal

aparcar, 2018-01-23:
@G10h4ck so extend watchping or pingcheck?

G10h4ck, 2018-01-23:
Evaluating which one is more adecuate can be part of the proposal, actually there was an idea about this last year

https://wiki.freifunk.net/Ideas_GSoC_2017#Improve_Easy_Internet_Sharing_in_LibreMesh

br101, 2018-01-23:
pingcheck can use HTTP (option protocol tcp, default tcp port 80) to check connectivity. it just checks if the tcp connect on port 80 succeeds or not, and does not request a webpage, though...

G10h4ck, 2018-01-24:
In that case it doesn't improve on the ICMP check, the problem with ICMP is that captive portals usually fake the ICMP answer so the checker thinks that there is internet when there is not, same happens with a simple connection on port 80...

One way to check this is doing like android phones does that checks for a specific google page that always return 200 OK

tekoholic, 2018-03-08:
Any progress on this? I can't seem to get watchping to do it's job. Not sure that pingcheck would either, am going to test it in the coming few days. In the end, which one I will support will depend on which one actually does what it is supposed to do. Another option for consideration might be apinger, although it is limited to icmp so far as I have read, but at least does seem to have multi-interface options, among other things that watchping simply lacks...

gmarcos87, 2019-03-06:
@tekoholic were you able to run any tests with conclusive results?

ilario, 2019-10-04:
See related #93 and [my comment on that issue](https://github.com/libremesh/lime-packages/issues/93#issuecomment-538345915).

ilario, 2020-01-09:
Is everyone ok with merging the discussion on #93? Feel free to re-open if you believe the two issues are different.

closed 2020-01-09 by ilario