libremesh/lime-packages #72: Remove luci-mod-lime-{basic/status}

p4u, 2017-02-19:
What are packages "luci-mod-lime-basic" and "luci-mod-lime-status" ? They seem to be old and placed on a wrong place of menuconfig.

Can they be removed?

altergui, 2017-02-23:
luci-mod-status is integral part of lime-full, and it's not old. can't be removed.
if it's on a wrong place of menuconfig, let's fix that

luci-mod-lime-basic is not actually being used, we could drop it, yes.
for future reference, it was a subset of luci-mod-admin-full, implementing just "Upgrade" (sysupgrade), "Backup/Restore config", "Change password" form, and "Reboot"

p4u, 2017-02-23:
So, do you have plans for doing it yourself? or should I do it ?

ilario, 2017-03-13:
This should be solved by #77

closed 2017-03-14 by ilario

altergui, 2017-04-04:
And the wrong placement of lime-mod-status was solved by 4a741e262a5690119d86f195c1c36d2b8bb84bf2