libremesh/lime-packages #75: lime-config: generate configs from scratch

p4u, 2017-03-11:
Copy from @dangowrt on lime-dev mailing list:

> lime-config currently re-writes UCI configuration. ie. if I mess up /etc/config/network, running lime-config will not necessarily fix it. There are also situations which make re-running lime-config impossible as that would break the config completely. 
> 
> To resolve that mess I suggest to refrain from re-writing UCI configuration and rather generate it from scratch every time by using the new /bin/config_generate script. 
> 
> If anything is missing in that new board-config aka. uci-defaults approach, then we should add it and discuss with upstream, ie. make sure that all board-specific details needed to generate the entire configuration are covered.
>

ilario, 2017-12-25:
Is the closing of #267 by #270 affecting this also?

gmarcos87, 2019-03-06:
#135 was closed a few days ago, does that mean this can also be closed or is it still a problem?

ilario, 2020-02-26:
@spiccinini could you give your opinion on this issue? Is this something that would help us? If I recall correctly from some discussion on #631, there was an easy way to do this, no?