libremesh/lime-packages #76: Splash page

amuuza, 2017-03-12:
It would be nice to have Nodogsplash or any other captive portal working. No authentication needed. Just to inform what the network is about.

FreifunkUFO, 2017-03-13:
Splash-pages are evil for net-neutrality, User-Experience and more.

There are better ways to inform users "where they are". you might name it with a nice SSID oder make (non-technical) banners and info-flyers.

panosnethood, 2017-03-13:
A possible idea to avoid splash pages is to create a trend of adding the "localhost" URL on the SSID itself perhaps with some special characters to signal that it is a "local" page?

FreifunkUFO, 2017-03-13:
The name of SSID cant enable or avoid a splash page.

Do you think about adding a SSID, only for the use of the splash-page (while all other wifis are still without splash?

panosnethood, 2017-03-13:
A, this could be also a nice idea, to have a separate SSID for the splash page. But what I was proposing is simply to create a habit of choosing SSIDs that contain the URL of the localhost, e.g., an SSID that would look like "My place --- http://localpage.here"

amuuza, 2017-03-13:
Thank you.

Why are splash-pages that bad? Why are they a problem on net-neutrality or user experience (beyond the first Accept click)?

In the lime-users mail list @p4u suggested an apparently clean solution with Tinyproxy.

In our case, just a meaningful SSID would not be informative enough. Writing a URL in the SSID seems a better solution, but there is no guaranty it will be visited by everyone, it is not such a powerful tool as the captive portal.

For our local community here, we consider a captive portal a key element for the success of the network. We would like to have, as a minimum, something like the following text written in the three main languages of the neighborhood. We may sometimes add important neighborhood news too.

"Welcome to... We are...
At this first stage of this community network we have two different kind of participants:
A) Those who cannot afford an Internet connection
B) Those who have an Internet connection and want to share it with A.
If you are A: Welcome!
If you are B but aren't sharing your Internet connection yet, please contact us, we will help you.
Contact details:..."

dangowrt, 2017-03-13:
Intercepting HTTP requests and redirecting them **is** problematic, for
two reasons:
a) it doesn't always work. some browsers start with HTTPS right out and
   then you are doomed, the user will see a security warning and recent
   browser will not even allow to make an exception. So unless some
   corrupted CA issues a wildcard cert for you (which is unlikely
   unless you are part of the so-called security industry or a
   government or pay a lot), you are doomed.
b) people may (and do) use IP-based networks for applications other
   then web-browser. think of a streaming-radio device with only a
   small display which cannot display HTML...

To at least go around probem (a) there is a sort-of clean solution to
that which works on most mobile devices and will cause them to redirect
you to a 'login is required' page even if you try connecting with a
random app rather than the browser. WISPr/UAM is used for that,
Coova-Chilli supports that and it doesn't require wildly redirecting
port 80 traffic.

p4u, 2017-03-13:
If I don't remember wrong, this modified version of tinyproxy [1] solves both points.

It only redirect the HTTP (port 80) traffic which is a WWW query (HTTP/1.0 or 1.1). So HTTPs traffic is never manipulated and the mobile APPS or other TCP/IP data is not captured.

[1] https://github.com/routek/qMp/tree/testing/packages/qmp-tinyproxy

FreifunkUFO, 2017-03-15:
@amuuza splash is not a key element for success. you have to rethink Layer8 problems. so even for marketing issues some flyers, info-posters, internet-webpages or commercials are better. for user information it doent make sense to tell the user "hello you are at ... and you are using this wifi-network" because he/she even knows the location and the wifi-name..

with freifunk we tried it 10 years before and finally were disabling it. only for networks without internet a splash-page would makes sense.. 

please see screenshots at https://wiki.freifunk.net/DHCP_Splash with almost same text like yours. 

https://wiki.freifunk.net/DHCP_Splash

or see http://doc.meshkit.freifunk.net/daily/html/usage/splash.html or https://forum.freifunk.net/t/router-in-dresden-ohne-splash-screen-betreiben/10848 unfortunatly all in german
https://forum.freifunk-muensterland.de/t/textentwurf-warum-vorschaltseiten-keine-losung-sind-warnung-lang/987/16 and https://freifunk-muensterland.de/warum-vorschaltseiten-keine-loesung-sind/

nicopace, 2017-03-31:
@amuuza could you share which is your use case?
Perhaps it doesn't sound reasonable or technically correct to implement it, but you may have a perfectly valid use case where this makes sense.

amuuza, 2017-04-06:
Thanks for all the replies.
If having a captive portal in my community network is going to be that problematic, I agree, I do not want that. I will have to accept it.
But, as I am not a developer, I can say I do think it is worth it to work on the development of a good, stable and smooth captive portal : )

I do not know how hard it is to achieve that, or if technically possible at all. But I keep thinking captive portals -if they worked ok- could be the key to expand community networks. it would be a very powerful tool, it would allow us to easily invite everyone to participate in the community. 

In our case, it would allow us to easily inform everyone what the network is about and would let us inform of current needs.

@p4u mentions a solution that might work (I do not dare to try it because I would need an easy step-by-step guide)

But that solution sounds ok to me. The default behavior should always be "permit", as the only goal of the captive portal would be to inform, not to authenticate. So, in case of problems, "permit".

But I am not a developer, maybe there are technical things I am not considering.

By the way @nicopace , I do not know what a use case is! Please let me know what more details you would need.

amuuza, 2017-04-06:
If the two problems @dangowrt described were solved. Would you @FreifunkUFO and @dangowrt use a splash page?
Would it be worth it to implement it in your cases?
Maybe there are other things you are considering and I am not aware of.
I guess one is the idea of forcing everyone to see a web-page every time they connect to the network.
When does that exactly happen? Does it happen every time a host receives a new DHCP address?

nicopace, 2017-04-06:
Hi @amuuza, I understand the concern... I think this is not about having this feature or not, but how it gets implemented and if it is a feature or a plugin (wether the feature is fixed in LibreMesh or is a feature that can be enabled/disabled).

A Use Case is a description of a need from a User's perspective.
Are you a member of a Community Network? Why do you think this is the way to go?

We are now working on a Captive Portal that will be on the border of one of our networks in Quintana, Cordoba, Argentina. The rationale on why we put this is to give the organization of the Community Network have a better control on who gets connected to the collective connection to the Internet that they are co-managing.

Please, share with us a little more about why you need what you are asking, and maybe we get to understand and better solve your issue.

amuuza, 2017-04-11:
Thank you @nicopage

I live in a district where there are many poor people. Most of them do not have an Internet access, and very likely that is not their priority. They have more basic needs.

Unfortunately the neighborhood lately started to be considered as a trendy one. Gentrification and touristification has just started with the help of some people and institutions, like the local government. They say their policies and investments will help everyone. But their policies and selective investments are just speeding up gentrification. Many people see now the neighborhood as a business opportunity. Some houses have started to get the Airbnb infection and rentals and services in the neighborhood started to go high too. The most vulnerable neighbors have started to suffer it. If things go on, sooner or later they will all have to leave the houses they live in and their neighborhood because they will not be able to afford it.

All those things do not happen accidentally. It is just a kind of colonialism. Many people will be very happy about it, not only because now loaded fashionable people will replace poor people in the new trendy neighborhood, but also because of racism. Many of the current neighbors are Romanian gypsies and Spanish gypsies, they both suffer daily exclusion and hate from some of the well-established white local neighbors. And often from police too.

On the other hand we have some grassroots activists who fight gentrification and racism. They campaign against every move the local government or any other organization does favoring gentrification. They also build community resistance through different self-managed activities involving the unprotected ones. Their goal is to defend and empower the most vulnerable population here, so that they can stay in the houses they live in now. They are organized through an assembly where they try to practice mutual aid.

We came up with this idea of building a free wireless community network long ago, but everything seemed easier when we knew about LibreMesh. The first goal of the network was to provide Internet access to those who cannot afford it, the second one is to contribute to get neighbors together. The different ethnic groups and some families are excluded, sometimes they do not have any relationship with the rest of the neighbors. Some of them do not even speak Spanish, the official language here. They may like to know their economic problems are not just an isolated case of bad luck.

In such a context, we thought a splash page will give us the opportunity to explain in Romanian, Valencià, Spanish and English things like:

1.- This network is, as a first stage, for those who cannot afford an Internet connection and for those who have an Internet connection and want to share it for free. We need people willing to share their Internet connection. (And contact details to get involved)

If you are a tourist, please disconnect. Be aware that we are suffering touristification in the district.

2.- Gentrification has landed in the neighborhood. If you cannot afford to live here anymore, you are not the only one. There is an assembly in the neighborhood where we all can help each other, build community and organize to fight gentrification. (And details on when and where the assembly is).

3.- Neighborhood news.

4.- Technical guide to communicate within the community (forums, messaging, etc.).
 
The network currently has only 4 nodes covering two areas of the district. Now we are about to deploy nodes in 3 other areas covering many people in need.

nicopace, 2017-04-11:
I'm super curious now... where is this?
I'm sure the voucher captive portal i'm writing will serve your purpose.
https://github.com/libremesh/voucher
It will be optional inside libremesh as it is not a requirement for libremesh to work... but you will be able to install it soon.

nicoechaniz, 2017-04-12:
I just wanted to add that the captive portal solution @nicopace is working on was a request from many communities where the situations have similarities to what you describe. It's great to know that LibreMesh is being deployed in these scenarios. Thanks for sharing.

amuuza, 2017-04-12:
It is in Valencia, Spain. The district is Cabanyal. You will just find confusing news in the media.
Grass-roots activists' points of view are at:
https://espaicabanyal.noblogs.org/
https://twitter.com/EspaiCabanyal
My next goal is to convince them to abandon centralized services. Like this one : )

amuuza, 2017-04-12:
Is there any approximate estimate on when this solution will be ready?

But, in my case, I do not see a need for a voucher system, at least for now. We would like to avoid any administration. We were just thinking of a static page and an "Understood" button.
Do you still think this voucher captive portal will fit our needs?

nicopace, 2017-04-12:
Awesome @amuuza 

The idea is to have it working by the end of this month.

If you just need a captive portal with automatic aproval.. that is something you can do just with NoDogSplash (NDS).

amuuza, 2017-05-22:
Any update on the topic?

I still would like to have Nodogsplash or any other captive portal working. No authentication needed. Just to inform what the network is about.

There is a new maintenance release of Nodogsplash [1]. Does anyone know how to make it work on Libremesh? And produce a step-by-step explanation...

Thanks.

[1] https://github.com/nodogsplash/nodogsplash/releases/tag/v1.0.2

nicopace, 2017-05-23:
Hi @amuuza,
We ended up implementing our own captive portal solution.
You can check it out over here: https://github.com/libremesh/pitbull
We are polishing it up, we will release it soon and will let you know.

amuuza, 2017-05-23:
Hi @nicopace,
Thank you. I'll wait.
How will I know it is ready? Will you announce it here? Will you announce it at lime-users@lists.libremesh.org?

Please let me know if you need to test it. I'll be happy to try it!

nicopace, 2017-05-23:
Will do here, and in the list.
Thank you @amuuza ... will let you know about the testing :)

patogit, 2018-05-12:
Our use case in rural Ecuador is a bit different. Right now, we think of the community network as a commons, and not as a free-for-all. (Maybe someday that will change, and anyone with a device can access the Internet for free, because our face-to-face coordination will function well, and maybe we'll have a free connection from some institution's extra "bandwidth".)

For now, seeing the Internet access as a commons means that it is managed, and there are norms about who participates and how. The network members don't want to share their Internet connection with people who don't support the network by participating in meetings, performing maintenance tasks, or contributing money.

This means that we would like a way to stop people who don't support the network from accessing it -- this mostly means people who come from nearby towns. We have been thinking about a captive portal and time-limited access codes. So someone would talk with the network managers to get a code that gives them Internet access for an hour, a week, a month, a year, or forever. The person would see a captive portal when they first connect, enter the access code, and use the Internet connection until the time expired. This way we can give network members access codes for a month, a year, or forever, and people from other places who don't support the network with their labor or creativity can pay for an hour or a week or access.

But right now, as I read the instructions for the only working captive portal access control solution I've heard of for LibreMesh -- http://wiki.coolab.org/index.php?title=How_to_create_a_captive_portal (a well written tutorial, thanks @brunovianna ) -- I think about it... implementing this and managing this so that it works smoothly would be challenging, especially since access codes aren't time limited. I think @FreifunkUFO 's point that there are non-technical actions that are more effective makes sense here. In this community, it's pretty easy to see who's near a WiFi zone and know who they are (there are only 45 residents, and most are family). So using a captive portal that only does part of what we want would be a headache for managing the commons. It makes more sense for use to find ways to manage our commons in other ways, and that might still be true even if there were really nice software for access control. The experience of having Internet access and managing it as a community might feel way better and easier and more human by creating a way to do it without access codes. Since that's the only option right now (or learn how to make the software, which is very unlikely), we'll do it and see how it goes for us.

brunovianna, 2018-05-14:
Hi Pato

> But right now, as I read the instructions for the only working captive
> portal access control solution I've heard of for LibreMesh -- [...]

Just to be sure, we always use time limit on the codes. They're set by
default for 30 days, but you can change it if you'd like.

-- 

bruno@pobox.com ▀─█▄██▄▀▄
http://brunovianna.net ─█▄██▄▀█▀█▄
skype: randomico▀─█▄██▄▀█▀█▄▌██─█▌█▌

patogit, 2018-05-14:
> Just to be sure, we always use time limit on the codes. They're set by default for 30 days, but you can change it if you'd like.

How? Is the expiration time hard-coded, or is it relative to when the code is used/activated? Do all the codes have to expire at the same time, or can some codes expire an hour after being activated, and others expire a month after being activated?

(Even if we decide not to use them in the network where I live, others might want this.)

brunovianna, 2018-05-14:
the system counts 30 days from the moment the user enters the code for the
first time.
you can see the timestamp recorded on the db.csv file.

line 16 on the vale.sh script sets the number of seconds after the first
use:
https://github.com/coolabnet/lime-packages/blob/develop/packages/vale/files/usr/bin/vale.sh

brunovianna, 2018-05-15:
> Could we modify vale.sh so that it reads another column from db.csv for
> each code that has the number of seconds that code is valid? Make two new
> columns in db.csv for vale_secs and warn_before_secs and change a few lines
> in vale.sh?

Yes, that is possible, but you can also get the number of valid seconds
just by subtracting initial timestamp from the current time at any moment
you need it :)

brunovianna, 2018-05-17:
> I don't want to know how long it's valid for, instead I want to set that
> code1hour is valid for 3600 seconds, and code1week is valid for 604800
> seconds. This way different people/devices have access for different
> amounts of time. For example, imagine a community center with WiFi that's
> always on, but there isn't always a person there to coordinate the space.
> Someone who wants Internet access can go to the coordinator's house, buy
> three codes for one hour each, and then use them throughout the week.
> Someone else buys a code for an entire week. Community network members get
> a one month code when they pay their dues.

yes, that makes perfect sense. some places here wanted to sell different
codes for tourists (3 days) and neighbors (30 days).

but for that you'll need more coding.  if you get it to work, let us know!

also, hiure wrote some a nice managing script for the vouchers:
https://github.com/coolabnet/VoucherAdmin-v2.0

ilario, 2019-10-04:
@gmarcos87 can [pirania](https://github.com/libremesh/pirania) be used also as a simple splash page (like a "welcome to my network" page) without any ticketing system?

ilario, 2020-01-09:
@luandro can pirania be used also as a simple splash page (like a "welcome to my network" page) without any ticketing system?

luandro, 2020-01-09:
Nope. That's a good feature to think about in the future, once we get this release stable.

There's a way to hack it to behave like that though. Create a voucher valid to a huge amount of macs, and hard-code on the front-end it's code.

ilario, 2023-03-03:
Is it possible to do this with Pirania now?
@luandro @spiccinini

luandro, 2023-03-03:
@ilario I believe this feature is still not built into Pirania yet.

henmohr, 2024-09-26:
Hey. Pirania with the option with_vouchers '0' will display a count down and when the users clicks the button it will add its mac address to the allow list. Maybe this can be used as a splash page? Just change [here](https://github.com/libremesh/lime-packages/blob/master/packages/pirania/files/www/portal/read_for_access.html)

ilario, 2024-11-11:
Thanks @henmohr !
To me it looks like a splash page. Closing. Anyone, please feel free to re-open explaining which feature is needed that Pirania does not have yet.