libremesh/lime-packages #100: Autoconfig SSID

nicopace, 2017-03-31:
Instead of setting up the name of the network on firmware-build time, allow the firmware to decide which network will be by looking at the networks that are close to him.
Give the user an interface in which the user can choose to join a network that is already deployed, or to create a new one.

aparcar, 2018-02-11:
isn't this covered by the firstboot wizard?

ilario, 2018-12-08:
This is very epic stuff, can't we consider this problem addressed by network-profiles + Chef?

aparcar, 2018-12-09:
I'm thinking of creating an interface to create network profiles online. Maybe integrated into chef

ilario, 2019-10-04:
As @aparcar wrote more than one year ago, this is the work of FirstBootWizard. Closing.

closed 2019-10-04 by ilario