libremesh/lime-packages #102: Fix 11s support on Atheros chips

nicopace, 2017-03-31:
In the current implementation of Atheros' 802.11s there seems to be a bug that hangs the kernel.
@p4u and @altergui may be able to add more detail to this issue.

dangowrt, 2017-03-31:
/me is going to pick up a serious amount of ath9k test gear from @FreifunkUFO at some point during the weekend

nicopace, 2017-03-31:
@dangowrt awesome!!! Let us know what you find out :)

FreifunkUFO, 2017-04-03:
which firmware?
kernel-hangs? so we might need serial console?

FreifunkUFO, 2017-04-03:
some freifunk-gluon users also uses 11s.. we have to ask them about their experiences and wifi-config.

https://github.com/freifunk-gluon/gluon

dangowrt, 2017-04-03:
Looks like @nbd168 just fixed 802.11s in LEDE, at least that kernel bug does't get triggered any more.
See [commit cffeb311 mac80211: backport an upstream fix for queue start/stop handling](https://git.lede-project.org/?p=source.git;a=commit;h=cffeb31107e27a48ddaf4953590acdefc66f2de9)

nicopace, 2017-04-03:
Are you sure @dangowrt? it doesn't seem to mention much more than 'Fixes issues with 802.11s'... we should test

dangowrt, 2017-04-03:
It looks like it, but I'm about to test this excessively just now...

FreifunkUFO, 2017-04-04:
ok, i found a ticket about this stuff: https://bugs.lede-project.org/index.php?do=details&task_id=644&order=dateopened&sort=desc

FreifunkUFO, 2017-04-04:
and i found another ticket: https://bugs.lede-project.org/index.php?do=details&task_id=619&order=dateopened&sort=desc

dangowrt, 2017-04-04:
#619 is the one which got fixed by the patch @nbd168 imported from
upstream (I referenced previously).\

#644 got an obviously wrong wifi config (option ifname is the most
likely cause of the probolem, coverage '5' seems wrong, too)

p4u, 2017-05-11:
Fixed in the last release

closed 2017-05-11 by p4u

FreifunkUFO, 2017-05-12:
so why there is a need for an extra (gluon?-)snippet/patch?
https://github.com/libremesh/lime-sdk/blob/master/snippets/patch_sdk_11s.sh
its uploaded by anonymious!? :-o

ilario, 2017-05-12:
anonymous = p4u, as he [confirmed](https://lists.libremesh.org/pipermail/lime-users/2017-May/000604.html) in lime-users mailing list

p4u, 2017-05-12:
That was a mistake. I commited from another computer where my name was
not configured properly on git.

On 12/05/17 16:41, Ilario Gelmetti wrote:
> anonymous = p4u, as he [confirmed](https://lists.libremesh.org/pipermail/lime-users/2017-May/000604.html) in lime-users mailing list
> 

-- 
./p4u

dangowrt, 2017-05-12:
See also https://github.com/libremesh/lime-packages/issues/123 which seems to be a duplicate of this bug for more details