libremesh/lime-packages #103: Add first_boot functionality to Reset Button

nicopace, 2017-03-31:
This was tested on WDR-3500/3600/4300 hardware with LibreMesh/OpenWRT ChaosChalmer firmware.

Expected behaviour: long pressing the reset button should reset to factory (fresh LiMe firmware).

Actual behaviour: nothing happens

ilario, 2017-05-25:
Does this work in LEDE? Is it a LiMe-only problem?

nicopace, 2017-05-29:
I don't remember well... but i think i tested with vanilla LEDE also

ilario, 2017-05-29:
[Here](https://lede-project.org/docs/user-guide/failsafe_and_factory_reset) on the LEDE wiki they refer to the reset button just for entering failsafe mode...

nicopace, 2017-05-29:
the thing was that originally, the stock software comes with this feature... so for flashed devices it would be interesting for the device to work in a coherent way with it. at least it happened to me and Gui that we were giving a workshop and actually expected for that to work and it didn't... but well... what do you think?

ilario, 2017-08-01:
I think that we have to wait for this to be solved on LEDE/OpenWrt side. Shall we close the issue here?

nicopace, 2017-08-01:
But if they don't solve it, is this a feature that we are interested in pushing it? this was an issue that arose in a workshop that @altergui and I gave in Mexico.

ilario, 2019-10-04:
I think it would be an interesting feature to have, but it makes no sense to solve it on a LibreMesh level as it is an OpenWrt thing.
Can we close?

closed 2023-03-03 by ilario