libremesh/lime-packages #104: If dnsmasq is choosen in menuconfig, lime-full disappears

nicopace, 2017-03-31:
@p4u and @altergui, can you expand on this?

G10h4ck, 2017-04-04:
This not a bug, is the expected behaviour, in openwrt a packet cannot say something like -dnsmasq, you can only express incompatibilities, some of the lime packages are broken if the dnsmasq package is installed and as limefull depends on them nor them nor lime-full appears, lime-build was the one in cvharge of removing problematic packets like dnsmasq

p4u, 2017-04-04:
It is not a bug but IMO we must find a way to do it differently. I see
two use cases where the current implementation originates problems:

1. It is very weird for a end-user to enter the menuconfig and do not
see the LiMe collection packages because dnsmasq is enabled by default.
Only those who knows that dnsmasq must be disabled would be able to
select LibreMesh.

2. If a user has a node with a standard LEDE installation and wants to
install LibreMesh trough OPKG, there will be a dependency conflict and
the packages might not be installed.

So, I would remove this -dnsmasq dependency and find another way to fix
the possible conflicts among dnsmasq-dhcp6 and standard dnsmasq.

On 04/04/17 11:41, Gioacchino wrote:
> This not a bug, is the expected behaviour, in openwrt a packet cannot say something like -dnsmasq, you can only express incompatibilities, some of the lime packages are broken if the dnsmasq package [...]

-- 
./p4u

dangowrt, 2017-04-04:
Maybe the best would be to implement lease-sharing via alfred to work with odhcpd? dnsmasq is PITA imho and the sooner we get rid of it, the better...

p4u, 2017-04-04:
On 04/04/17 18:35, Daniel Golle wrote:
> Maybe the best would be to implement lease-sharing via alfred to work with odhcpd? dnsmasq is PITA imho and the sooner we get rid of it, the better...
> 

I agree, we must get rid of dnsmasq.

Actually we are already using ALFRED for lease sharing. But IMHO the
best would be to use bmx6/7 SMS plugin instead. ALFRED does not work as
well as we would like to (according to our experience).

@Gui is our dnsmasq guy, would be interesting to know why are we
actually using it. So we can see if odhcp + some-tiny-dns-server would
be enough.

-- 
./p4u

dangowrt, 2017-04-04:
Hm, I do believe alfred is the right place for it, as leases need to be shared within a layer-2 domain (ie. batman-adv) and there is no sense to share them on layer-3... There are still some issues with batman-adv if we got a setup where we got nodes on different channels/bands connected via ethernet and hence they are on the same layer-2 network but may not be able to do multicast or exchange topology/originator info via batman-adv. having both, plain unencapsulated frames in one VLAN and batman-adv in another VLAN would be desirable to fix this, but as mentioned there was a race-condition in BLA which made this setup too unreliable. I remember we discussed this with @simonwunderlich in Porto...

p4u, 2017-04-04:
On 04/04/17 18:51, Daniel Golle wrote:
> Hm, I do believe alfred is the right place for it, as leases need to be shared within a layer-2 domain (ie. batman-adv) and there is no sense to share them on layer-3... There are still some issues [...]

bmx6 SMS plugin can also be used to share leases only in the layer-2
domain cause every node can choose which SMS wants to get (on-demand) so
we can use the "ap_ssid" (which is also used to calculate the bat-adv
VLAN) to generate a name for such SMS, such as leases_$ap_ssid". Then
let lime-config autoconfigure bmx6 to get "leases_$ap_ssid" SMS.

In my experience bmx6 SMS plugin works pretty well, the changes are
transmitted immediately (when file changes INOTIFY triggers bmx6),
partially (only the changing part) and in a very efficient way.

In ALFRED to transmit the changes in a file we need a cron rule which is
executed every 5 minutes which tels ALFRED to retransmit the file. It is
kind of annoying cause there is this 5 minutes window. I know we might
implement a C code to subscribe via (I/D)NOTIFY to the leases file and
automatically execute the ALFRED command. But BMX6/7 already implements
this feature and in addition we might remove ALFRED from libremesh, so
it is a piece of software less.

Maybe the best would be to implement it for both systems (bmx6/7 and
ALFRED) and leave it prepared in two different packages. So if some
community does not use bmx6 but bat-adv they can still use the lease
share trough ALFRED. But bmx6/7 is more desirable.

-- 
./p4u

nicopace, 2017-04-06:
@p4u didn't you said that sms bmx6 made bmx6d crash?

p4u, 2017-04-12:
There use to be a bug with bmx6 that made it crash when mixing nodes in the same network with and without the SMS plugin. But it seems tat it was fixed long ago.

G10h4ck, 2017-05-16:
I am not favourable to use bmx* plugin as we are introducing an hard dependency on an L3 protocol, while we are trying to solve a "same collision domain" L2 problem, the solution to this should be L3 routing protocol agnostic

p4u, 2017-05-16:
If L3 bmx protocol is installed, it can solve the problem by using SMS. So we don't need ALFRED.

If L3 bmx6 is not installed, ALFRED might be used.

So Alfred can by a conditional dependency, only installed and used if bmx6 is not installed.

closed 2019-03-14 by gmarcos87

ilario, 2019-03-15:
@gmarcos87 has this issue been solved? It looks like the [problem is still there](https://github.com/libremesh/lime-packages/blob/master/packages/lime-proto-anygw/Makefile#L24)

gmarcos87, 2019-03-15:
I think it's not an issue to solve, it's a condition not to have dnsmasq installed for libremesh to work properly. That in two years has not been resolved (especially user feedback) or generated a pull-request with an alternative is an example of this. In any case we could place particular emphasis on the documentation (in lime-web) on this issue.