libremesh/lime-packages #105: Make LibreMesh x86 build

nicopace, 2017-03-31:
It is useful for border routers that may not be wireless routers, or supernodes.

ilario, 2017-04-23:
Isn't this already possible?
In lime-build there's a x86 target defined [here](/libremesh/lime-build/blob/develop/targets/x86).
In lime-sdk there's a x86 and a x86/64 targets defined [here](/libremesh/lime-sdk/blob/develop/targets.list#L9).
I don't know if they can be selected in Chef.

nicopace, 2017-04-23:
I've done a LiMe build on x86 and it works.
As I understand the idea was to publish an official x86 image.
Chef now is an outdated tool that needs to be updated to use lime-sdk, so let's leave it for now.
I think @nicoechaniz and @p4u can add info on this.

ilario, 2017-05-25:
Can we close this issue and open one with focus on Chef upgrade?

nicopace, 2017-05-29:
@ilario Chef has his own roadmap, and is in it's repository.
I guess this issue is more related to having a x86 image published on the site rather than being able to generate an image (that we already can).

ilario, 2017-05-29:
You mean like [these ones](http://downloads.libremesh.org/community_chaos/16.07/x86/generic/) and [these ones](http://repo.libremesh.org/lime-17.04/targets/x86/64/Default/lime_default/)? If so they are already published, no?

nicopace, 2017-05-29:
It seems so... don't know really... would be great to know @nicoechaniz or @p4u ?

ilario, 2017-08-01:
I think we can close this, x86 images are there and works, do you agree @nicopace ?

nicopace, 2017-08-01:
Yes we can :) here they are: http://repo.libremesh.org/current/targets/x86/64/Generic/lime_default/

closed 2017-08-01 by nicopace