libremesh/lime-packages #107: Alfred automatic master node definition

nicopace, 2017-03-31:
In a BATMAN network that uses Alfred, you need at least one Alfred's master node, many is better (but not too many).
Define an heuristic and implement a way for this to happen.

ilario, 2017-08-01:
What happens currently? Are them all master or all slaves?

nicopace, 2017-08-01:
@nicoechaniz ?

ilario, 2017-09-10:
Currently they're all master. The decision for being master or slave could be based on the fact of being a gateway or not, which is a completely arbitrary factor but works IMHO.
I'm not sure about where to place the `alfred --modeswitch` commands for going slave or master, the easy way would be to include them in `bmx6-auto-gw-mode`'s `wan-ok.d` and `wan-fail.d` directories.

nicoechaniz, 2017-09-11:
Illario, that is exactly what we are doing manually in many networks. Setting to master the nodes that are gw. It is arbitrary but it makes sense because you usually build the network taking special consideration in having good path to a gw from every node, so this decision will optimize alfred's performance too.

nicopace, 2017-09-12:
one question regarding this.Ir a network doesn't have an uplink (a disconnected network), how will it behave?Can we add that if there is no uplink everyone will get to be master, and if there is uplink, all turn into slaves but the one that has the link? (@ilario moved my comment from #220 cause it was generating noise there... sorry)

ilario, 2017-09-12:
@nicopace ok, so the proposed approach does not work...
Or has to be completed by an additional check regarding the existence of at least an alfred master or about the total number of nodes and number of alfred masters. No idea how to do this...

nicopace, 2017-09-13:
So... no matter which routing protocol we use, we can now if we have access to internet or not, and whether the node that gives internet has an alfred-master node or not.
Using those two checks, we can decide if the node needs to be master or not.

ilario, 2019-03-15:
@gmarcos87 I suppose this has been closed as alfred is going to be replaced by shared-state? Can you confirm? Anyway [alfred is still dependency of lime-basic via dnsmasq-lease-share and dnsmasq-distributed-hosts](https://github.com/libremesh/lime-packages/blob/master/packages/lime-basic/Makefile#L28-L29).

gmarcos87, 2019-03-15:
I'm sorry I thought the changes in #476 modified the dependencies. We can leave it open until that happens, which I think is a matter of a couple of days.

nicopace, 2019-05-29:
we can close now, as that has been fixed.

closed 2019-05-29 by nicopace