libremesh/lime-packages #108: Remove any-gw functionality on interfaces connected via ethernet

nicopace, 2017-03-31:


G10h4ck, 2017-04-04:
Asi explicado no estoy del todo convencido, no veo claro como implementarlo y las implicaciones...

nicopace, 2017-04-04:
@nicoechaniz could you give @G10h4ck context for this issue?

p4u, 2017-04-04:
@Gui knows better about this issue, but the problem is mainly created by
the physical switch of the network devices such as Tplink 3500/3600/4300.

When using anygw in Ethernet and two nodes are connected through cable,
the same MAC address (anygw) is shared among both nodes. The CPU of the
switch stores a table with the relation MAC<->PORT. Then the state of
this table is changing constantly depending if the packet comes from one
node or from the other (depending on the Ethernet port). If this change
was done immediately, this behavior would not be a problem. But it seems
that most of these switch CPUs have a Time window (something close to
20s) where a new change for the same MAC cannot be done. Then it is
actually a problem and the consequences are very unpredictable.

In any case, disabling anygw from the ethernet port does not seem have
important implications for the whole network architecture. Usually when
a client is connected through cable it does not need to roam. The only
corner case I can imagine is when using stupid APs (not libremesh
bridged AP-Ethernet device) connected to a Ethernet port. In this case
the roaming would not perfectly work among these stupid APs if connected
to different libremesh nodes. But IMO we might accept this issue.

I hope it is more clear now.

On 04/04/17 17:23, Nicolás Pace wrote:
> @nicoechaniz could you give @G10h4ck context for this issue?
> 

-- 
./p4u

p4u, 2017-05-30:
I've recently experienced the problem described above in a small setup of LibreMesh. An option would be to add a new config boolean option "anygw_only_wifi" to enable or disable AnyGW on the Ethernet ports. So at the end user (or community administrator) can choose. @G10h4ck, do you see it feasible?

G10h4ck, 2017-05-30:
Still this bug miss the context as the explanation you provided is about another bug, adding an option is not a problem, thought I still miss the context where this happens (if it is a setup we support) and It is not clear to me how to disable anygw on specific interfaces as it is built on top of br-lan

p4u, 2017-05-30:
This issue is not about a bug, is about a change on the libremesh network architecture. The issue I'm addressing is one of the points which gives sense to this possible change (considering that anygw involves also anyMAC and having multiple routers sharing the same MAC connected to the same physical switch is what generates the problem described). So IMO it makes sense to have this discussion here. However if you think that another issue with another title makes more sense for having this discussion, let's just move there. 

> and It is not clear to me how to disable anygw on specific interfaces as it is built on top of br-lan

Is not clear to me either, but the issue is annoying. We might create a second bridge (br-wlan) for anygw in case the options in enabled.

p4u, 2017-06-01:
Reading again the whole discussion, I see that we are misunderstanding each other. The more correct topic would be "Remove anyMAC from Ethernet devices" or even better "Use different anyMAC addresses on the Ethernet ports". However the possible solution is still not clear due the usage of br-lan as a single LAN bridge interface. 

Let me try to explain again the problem: it comes when there are two or more LiMe nodes connected to the same Ethernet switch, as they share same MAC, when a client connects to this switch, the annoying behaviour starts and the router switch performs weird (changing from one node to the other and breaking the TCP connections (broken pipe)).

Thinking loud: what would happens if the anyMAC is disabled but the anyIP is still used? When a client ask for the ARP to resolve the anyIP, the first node receiving the packet would send the reply and it would be used as gateway. But when the client roams to another node, would the ARP table be updated? I think it might depend on the client (how it manages the WiFi roaming). Would it make sense to be implemented by setting to 1 some config option like "disable_anyMac".

p4u, 2017-06-08:
So I vote for having an option on libremesh config to disable anyMAC in all networking devices. @G10h4ck what do you think?

p4u, 2017-06-09:
Push request sent #145

nicopace, 2017-06-11:
@nicoechaniz

ilario, 2017-09-25:
#145 was closed with @p4u's comment: 
"I don't think we should accept this change but instead we should find why the TCP connextions are broken (TCP reset) when the internal switch changes from one port to another."
Seems to me that also this issue needs to be closed, right?

ilario, 2018-12-08:
No comments in the last year after asking for closing, I suppose it has been fixed or is a very low priority issue? @p4u @nicoechaniz @G10h4ck 
Closing.

closed 2018-12-08 by ilario