libremesh/lime-packages #109: Measure performance impact of having multiple virtual AP

nicopace, 2017-03-31:
The current LiMe implementation uses many virtual AP (like multiple SSIDs). We want to know if this is a performance hit in order to take better decisions on how we use this functionality.

aparcar, 2018-12-09:
@dangowrt you know something about that?

ilario, 2019-10-04:
To know if the presence of multiple interfaces (and nodes or clients connected to it) affects the performances would be very useful also for discussing the optimal setup in #262.

nicopace, 2019-10-08:
I don't feel this is a release-blocker @ilario. could you expand or remove the tag? thanks!

ilario, 2019-10-08:
Sorry, I originally added the tag (and I removed it now) just because is a thing that so easy to test and potentially so important for the performances that I thought we absolutely have to do it before the next release.

spiccinini, 2019-10-08:
@ilario do you have an idea of how to measure performance?

ilario, 2019-10-08:
Meh, the idea was to set up a test network, maybe 3 routers connected in a line like A-B-C (I have only omnidirectional routers so maybe I would have to decrease the output power in order to avoid a direct A-C connection), with a client connected via Wifi to B (doing a little of background traffic) and then run an iperf3 between one PC connected by cable to A and another connected by cable to C.
The situations I would compare are:

A(2.4GHz: ap, apname; 5GHz: ap, apname, mesh)
B(2.4GHz: ap; 5GHz: ap, apname (client connects here), mesh)
C(2.4GHz: ap, apname; 5GHz: ap, apname, mesh)

A(2.4GHz: ap, apname; 5GHz: ap, apname, mesh)
B(2.4GHz: ap, apname (client connects here); 5GHz: ap, mesh)
C(2.4GHz: ap, apname; 5GHz: ap, apname, mesh)

A(2.4GHz: ap, apname; 5GHz: mesh)
B(2.4GHz: ap, apname (client connects here); 5GHz: mesh)
C(2.4GHz: ap, apname; 5GHz: mesh)

A(2.4GHz: ap, apname; 5GHz: mesh)
B(2.4GHz: apname (client connects here); 5GHz: mesh)
C(2.4GHz: ap, apname; 5GHz: mesh)

ilario, 2020-11-27:
Some testing also on this has been reported here:
https://lists.libremesh.org/pipermail/lime-users/2020-November/001817.html

ilario, 2020-11-30:
Closing, until when there will be hints of this being important.

closed 2020-11-30 by ilario