libremesh/lime-packages #110: Fix batman-adv Status WebUI

nicopace, 2017-03-31:
Expected behaviour: see nodes

Actual behaviour: nodes are not being displayed

FreifunkUFO, 2017-04-04:
its related to https://github.com/libremesh/lime-packages/issues/73 , unfortunatly @p4u was closing that ticket. 
there might be some more luci problems inside webui

i assume its this package:
https://github.com/libremesh/lime-packages/commits/develop/packages/luci-app-batman-adv

p4u, 2017-04-04:
@FreifunkUFO separated issues should be treated in different tickets. So
this is why now we have this issue opened with the specific topic name.

On 04/04/17 17:30, FreifunkUFO wrote:
> its related to https://github.com/libremesh/lime-packages/issues/73 , unfortunatly @p4u was closing that ticket. 
> there might be some more luci problems inside webui
> 
> i assume its this package:
> https://github.com/libremesh/lime-packages/commits/develop/packages/luci-app-batman-adv
> 

-- 
./p4u

nicopace, 2017-04-06:
I'm on it.
It appears that the issue the output of that 'batctl o' and 'batctl gwl' has changed, and that is why the interface is not showing anything.
Have to see how it changed and updated it.

ilario, 2017-04-23:
Seems related to (closed?) #1

nicopace, 2017-04-23:
Don't know... i've already implemented part of this.  @p4u said that @altergui had some word on this, as I could just implement one part of the change (because I have no access to a batman-adv network that generetes all the outputs that were shown there) and there are still some things to do to get this done.

If you want to see what was done, you can check it out over here:
https://github.com/libremesh/lime-packages/pull/113

nicopace, 2017-04-25:
This shouldn't be closed, as the fix I sent only fixed part of the issue.
I had no environment to test the Gateways section, just the Originators one.

p4u, 2017-05-25:
Github closed it automatically because I published a commit with the words "Fix #110". But I did not close it in purpose. 

Anyway, sorry.

ilario, 2017-08-01:
This is a duplicate of #1 (also closed by error).

ilario, 2017-09-12:
Closed by #255

commit 0e350d87 (2017-04-25): Fix batman-adv Status WebUI. Closes #110.
closed 2017-09-12 by ilario