libremesh/lime-packages #111: DNSMasq crashes and never gets up again

nicopace, 2017-03-31:
# Expected behaviour

After a DNSMasq crash, procd should get dnsmasq up again.

# Actual behaviour

@nicoechaniz saw that it is not getting up again.

I guess @altergui has been working on it.

One proposed workaround could be to replace dnsmasq with udhcpd and a tiny dns replacement. There are some replacements in https://wiki.openwrt.org/doc/howto/dns.overview

p4u, 2017-04-04:
http://lists.infradead.org/pipermail/lede-dev/2017-March/006828.html

ilario, 2017-08-01:
This should be fixed by Smonit merge in #142 and #177

nicopace, 2017-08-01:
@ilario have you checked this?

ilario, 2017-08-01:
No, actually my assumption was wrong. Smonit up to now just checks if dnsmasq is running as is commented in #142 by @G10h4ck.
Do someone have experience with this? How can this be checked by smonit?

nicopace, 2017-08-01:
@ilario I believe would be healthier if instead of closing an inactive issue, ask for feedback or test it. That way we prevent loosing issues in the issue bushes :)