libremesh/lime-packages #91: Safe Reboot Web Interface

nicopace, 2017-03-31:


ilario, 2017-04-23:
A bit of description please, what is this issue about?

nicopace, 2017-04-23:
Sure.
Some time ago @altergui developed a functionality called 'safe-reboot'.
The idea is to add a luci module that allows GUI users to use this functionality.
The way we thought about this was to add a button next to 'save and apply' like 'save and test' that would restart the router and apply this configs... but if the user does not go inside the router and says 'this config is valid' the router would restart itself after some time with the previous config applied and the new config pending as it were before.

ilario, 2017-04-23:
Ok, thanks.
Is this thing already implemented in LEDE?
If it's not, shouldn't we push things there?

---

> @altergui developed [...] GUI users
LOL

nicopace, 2017-04-23:
I don't really know what it takes to make this things implemented in LEDE... the luci module depends on safe-reboot... if safe-reboot is already on LEDE then having this on LEDE would be the way to go.

ilario, 2018-12-08:
web ui stuff... tagging @gmarcos87

gmarcos87, 2019-03-06:
Ok, related pull-request open in #477

nicopace, 2019-05-29:
This issue is outdated as we are not using luci anymore and this was thought from a luci perspective.

closed 2019-05-29 by nicopace