libremesh/lime-packages #92: Captive Portal with Tickets for accesing to Internet

nicopace, 2017-03-31:
Add the chance to manage the access to Internet through Tickets.
Add a Captive Portal that traps web browsers while users have no ticket, and allow them to use a ticket to remove the trap.
Add a web-ui to manage the tickets and the interaction with the Captive Portal.

nicopace, 2017-03-31:
Check these repositories:
* https://github.com/routek/qMp/tree/testing/packages/qmp-tinyproxy
* https://github.com/libremesh/lime-packages/tree/sandbox/vale-mulukuku

nicopace, 2017-03-31:
Nodogsplash communicates with the app that operates de bouchers via commandline. These are the calls it does:
  * <boucher binary> auth_voucher <client_mac> <boucher>: this is called after the user entered a boucher in the captive portal window, and is used to validate the boucher.
  Valid responses:
    * <seconds_authorized> <upload_limit> <download_limit>
    * seconds_authorized <= 0: boucher rejected
  * <boucher binary> auth_status <client_mac>: called before captive portal is displayed, may allow the user to browse before showing captive portal (just by showing client_mac).
  valid responses:
    * <seconds_authorized> <upload_limit> <download_limit>
    * seconds_authorized <= 0: captive portal will be shown

nicopace, 2017-03-31:
Nodogsplash code is here:
* https://github.com/openwrt-routing/packages/tree/master/nodogsplash
* https://github.com/nodogsplash/nodogsplash

nicopace, 2017-04-01:
First draft in https://github.com/nicopace/lime-packages/commit/bd6d2c5e319a7667373b1cae818ba358b0002082

ilario, 2018-12-08:
I think this is being implemented in pitbull, right? @nicopace 
Is it complete? Is it thought for being included by default in full LibreMesh images?

closed 2019-10-04 by ilario