libremesh/lime-packages #93: ForceWAN Web Interface

nicopace, 2017-03-31:
A user wants to configure what happens when he has WAN access. The options would be to force share, force not to share, or let the algorithm choose what to do.

In spanish: wan autodetecte que tiene internet, o que lo fuerces a que no o a que si.

nicopace, 2017-03-31:
@p4u @altergui is the isue well defined? I got this from a postit on the board today and have failed to get a clear idea of what it ment.

ilario, 2017-05-25:
I think it is clear enough. An example: you plug internet into WAN port of your LiMe router and use WiFi from LiMe router for accessing internet in your house, but you don't want your internet to be shared with the rest of the LiMe network. Or even you plug in but by default you don't want to use it for accessing the internet (e.g. because or erratic connection performances due to ISP).

altergui, 2017-09-14:
This was actually misunderstood, sorry.
The original idea when we wrote the postit is this:
a user connects a LiMe router WAN port to a ADSL, but the ADSL provider filters out ICMP pings. (I have personally encountered several cases like this). Everything works fine, besides ping. Thus, watchping doesn't realize there's "Internet" connectivity, and doesn't trigger the bmx6 tunIn announcement.
Resulting behaviour reported by the user: The LiMe router gateway provides Internet access (if you directly connect over wifi to it) but it doesn't "share" it with the rest of the mesh (i.e. when you connect to any other node)

This would be solved by a "simple checkbox" in the web interface which basically tells a router "Please announce that you have Internet connectivity to the rest of the mesh, even if you can't detect it with your poor watchping method"
Behind the scenes, it would basically force the tunIn announcement in bmx6 (just like watchping does)

ilario, 2017-09-22:
Ok, so the problem would be mitigated also with a solution like replacing watchping as discussed on #66.
In my opinion both what you wrote and what I wrote are worth to implement, they would fit in two different radio buttons, something like this, in the webui:

Use directly connected gateway:

- always, without checks
- depending internet availability
- never

Share direct internet access with other nodes:

- yes, allow other nodes to use this as gateway
- no, just clients of this router can use it as gateway

ilario, 2018-10-26:
Coming back to the original scope of the issue: announce internet access without checking, some ways to do this:

- add a new entry in /etc/config/lime* system section for configuring the destination IPv4 and IPv6 for watchping, then this information gets written in /etc/config/system. For having always a positive it would be enough to set `127.0.0.1` as IPv4 target and `::1` for IPv6 target.
- add a (bmx6|bmx7|batman-adv)-auto-gw-mode option in /etc/config/lime* that gets checked in `bmx6-auto-gw-mode/files/etc/watchping/wan-fail.d/bmx6-gw` but this would send anyway an useless ping
- add a `watchping_enable` option in /etc/config/lime* system section that activates/deactivates watchping.
- somehow disable watchping from init

For the last two options, would the wan be always shared or some commands (like the ones [here](https://github.com/libremesh/lime-packages/blob/master/packages/bmx6-auto-gw-mode/files/etc/watchping/wan-ok.d/bmx6-gw)) have to be issued?

ilario, 2018-10-26:
For example, in `/etc/init.d/watchping` something like [the solution for AutoAP](https://github.com/libremesh/lime-packages/blob/master/packages/lime-ap-watchping/files/etc/init.d/auto-ap#L12) (I actually don't like it: it processes lime* config files out of lime-config) could be added, looking for a watchping_hosts in /etc/config/lime* in the network section (I suppose this will go to the network section as autoap_hosts is also there, and is more or less the same thing).

More elegant would be to add a "openwrt" section in /etc/config/lime* files with relative openwrt.lua file for editing generic openwrt options not included in the other sections?
The resulting config file would be something like this:
```
config lime system
    option ...
config lime network
    option ...
config lime wifi
    option ...
config openwrt system
    option watchpingSPACERpinghosts 8.8.8.8
```
where SPACER could be more or less whatever special character that uci likes (in the option names there's usually no special character, so any of them should be ok, for example we can safely use here the colon `:` which was problematic in #409).

ilario, 2018-12-08:
On #403 @gmarcos87 pointed out that announcing a non-working internet connection is a very bad idea as Android's apps does not like it at all.

gmarcos87, 2019-03-06:
The idea is then to add a configuration option that is: This node is a gateway, announce it to the mesh. I'm thinking it can be problematic, someone can break connectivity just by announcing false connections.

ilario, 2019-10-04:
> a user connects a LiMe router WAN port to a ADSL, but the ADSL provider filters out ICMP pings

To address this case we should just improve watchping to download a small file from some reliable server (like https://fedoraproject.org/static/hotspot.txt or http://network-test.debian.org/nm or http://networkcheck.kde.org/ or http://start.ubuntu.com/connectivity-check.html or the empty http://connectivity-check.ubuntu.com) rather than using ICMP ping (or replace watchping with some better tool).

Anyway, Babeld (which will replace BMX6 in the next 19.something release) just announces all the default routes it gets, without checking in any way, correct?
@spiccinini @G10h4ck @nicopace @gmarcos87 @altergui should we create a new package `babeld-auto-gw-mode` (and improve or replace watchping as described above)?

ilario, 2020-05-05:
> Babeld [...] just announces all the default routes it gets, without checking in any way, correct?

Confirmed. If a default route is received by the router, Babeld does not check if it works.

> should we create a new package babeld-auto-gw-mode

I think that we can do something even better than an on/off switch: in case of wireless WAN connections, the quality can be estimated and communicated to Babeld as a metric (or something else? See Babeld man page for more options: https://www.irif.fr/~jch/software/babel/babeld.html ).
The metric can be set with:
```
uci set babeld.default4.action="metric 12345"
/etc/init.d/babeld reload
```
or maybe (I didn't test) also with `babeld -C ...` command.

spiccinini, 2020-05-26:
Minimal `check-internet` script implemented in #712

spiccinini, 2020-12-01:
Babel is not in charge of checking if the routes published by the kernel are good or not, it assumes that they are.
Verifying that the routes are good should be done with other protocols.
In the case that the provider is simply a DHCP, Julius wrote a little tool called babel-pinger.
Basically what it does is use a loopback interface and publish and unpublish the route if a configurable echo / dns / ntp server stops responding.
If we want to use that, we must change so that the wan route is not published, only the loopback route (or something like that).

babel-pinger package in openwrt https://github.com/openwrt-routing/packages/blob/master/babel-pinger/Makefile
Other useful links about this small software: 
* https://www.irif.fr/~jch/software/babel/babel-pinger.html
* https://alioth-lists.debian.net/pipermail/babel-users/2009-May/000261.html
* https://alioth-lists.debian.net/pipermail/babel-users/2008-September/000160.html

amuuza, 2020-12-01:
I think Libremesh is supposed to make sharing resources an easy and by-default thing. When this issue was open, Libremesh was doing so, and it was open just to fix a specific case:[ when an ISP filters ICMP traffic](https://github.com/libremesh/lime-packages/issues/93#issuecomment-329533647).

I wonder if, while looking for a solution for that specific case, it would be possible to go back to the original state, when Libremesh was sharing resources by default. Now it is not sharing Internet connections at all.

amuuza, 2020-12-01:
Well, I guess I am missing some complications I am not aware of, so apologies if my previous post did not add any valuable information. I just would like to have that feature back!

ilario, 2020-12-04:
Sorry for not reporting the content also here but I threw a couple of ideas in this email: https://lists.libremesh.org/pipermail/lime-users/2020-November/001828.html

In order to detect if a node has internet access we can use the [check-internet package](https://github.com/libremesh/lime-packages/tree/master/packages/check-internet).

Both for ensuring that this runs only on the nodes which actually have a WAN connection and for executing corrective actions, we can use [watchping](https://github.com/libremesh/lime-packages/tree/master/packages/watchping) extending it with the configurable option to use check-internet instead of ping. Alternatively, we can copy the watchping package with a new name (e.g. watch-check-internet) and just rewrite the needed parts.

What to do in case of working/non-working internet is more dubious.

We could instruct Babeld to stop sharing the broken internet access to the rest of the mesh, like it was done in the previous releases, see for example in [bmx6-auto-gw](https://github.com/libremesh/lime-packages/blob/master/packages/bmx6-auto-gw-mode/files/etc/watchping/wan-ok.d/bmx6-gw);

One small problem I see with this is that anyone directly connected to the gateway LibreMesh node will still be using the broken internet connection, no? I mean, the other nodes will stop receiving the broken route but the very node connected via WAN to the internet (let's call this node "the gateway") will still use the broken route, no?

An idea to solve this: in the case that Babeld can share the route from all the gateways to all the gateways (I don't know if this is already is the default...), each gateway can evaluate its own connection and increase its metric (i.e. decrease the priority) and Babeld has an option to consider this value ([reflect-kernel-metric](https://www.irif.fr/~jch/software/babel/babeld.html#Global_options), did not manage to have it working yet). Once the metric has been modified, the watch-check-internet script should make sure to check using the local WAN interface (this could be done with curl's `--interface` option or wget's `--bind-address` option (not implemented in uclient-fetch, so we should include the official wget)).

PR #844 (2021-02-25): RFC add babeld-auto-gw-mode
By default babeld will redistribute all the routes installed even if they "don't work". For example when the internet provider use DHCP and the service is not working but interface is up, the route is installed but not working and babeld will anounce the non working route to the network.

This package provides a solucion using watchping hooks that adds routes with a special protocol number (7) when the WAN port has a working internet access and removes this route when the internet connection is not working as detected by watchping.

Fixes  #93 #822

Open questions:  [... cut]
closed 2021-02-25 by spiccinini