libremesh/lime-packages #95: Add support for choosing upstream DNS

nicopace, 2017-03-31:
When connected to an upstream connection, the DHCP server gives you DNS information.
Implement a way to choose it instead of a fixed one.

nicopace, 2017-05-21:
que todos los nodos de una nube resuelvan a traves del DNS que obtuvo el nodo gateway por DHCP. Pero que tambien haya la posibilidad de usar los que estan en /etc/config/lime

ilario, 2017-08-01:
We could even use the one suggested by the ISP by default, why do we set different ones?

ilario, 2017-09-22:
I tested that it's possible to use ISP upstream DNS servers just un-setting the custom ones, so replacing

```
        list resolvers '4.2.2.2'
        list resolvers '141.1.1.1'
        list resolvers '2001:470:20::2'
```
with
```
        list resolvers ''
```
in `/etc/config/lime` and running `lime-config`.
Just deleting the resolvers list from `lime` file doesn't work as it gets re-generated, to set it empty works.
I verified that the DNS server is the upstream one looking at `logread | grep nameserver`.
This needs to be documented, at least in lime-example, and implemented in web-ui.

ilario, 2018-02-04:
Fixed by @aparcar in #284

closed 2018-02-04 by ilario