libremesh/lime-packages #96: Review IPv6 autogw mode

nicopace, 2017-03-31:


p4u, 2017-05-16:
More info here: https://github.com/libremesh/lime-packages/pull/71

Added @dangowrt as assignee, cause he sent the first push request.

dangowrt, 2017-05-16:
I was played a lot with isc-dhcp lately and hence had a deeper look at how dynamically assigned IPv6 addresses and prefixes are being propageted and delegated using ubus. Imho this be implemented in such way that a change in the assigned IPv6 address/prefix from upstream (ie. xDSL or Cable ISP) will trigger a notification to change the announced gateway prefix in bmx6. I can take care of that.
However, what is still missing is the functionality on the client side which requiress bmx6 to dynimically assign the current prefix belonging to a gateway to its tunIn interface.

p4u, 2017-05-16:
> However, what is still missing is the functionality on the client side which requiress bmx6 to dynimically assign the current prefix belonging to a gateway to its tunIn interface.

Then would be nice to see if @axn has the time to implement this on bmx6.

G10h4ck, 2017-06-29:
We talked about this in Athens and we decided this will not be included in for this release

closed 2017-06-29 by G10h4ck