libremesh/lime-packages #97: Bundle Lime Workarounds

nicopace, 2017-03-31:
During the development of LibreMesh we found several way to deal with upstream inconcistencies or hardware failures. These ones we called them worarounds.
Get all of them together into a package to easen its maintenaince.

G10h4ck, 2017-04-04:
Hardware specific quircs are supposed to be handled with hardware detection plugins

ilario, 2017-05-25:
Are the workarounds these ones?
https://github.com/libremesh/network-profiles/tree/master/quintanalibre.org.ar/comun/etc/uci-defaults

nicopace, 2017-05-29:
@nicoechaniz ?

nicopace, 2017-05-29:
I guess so... but let's hope others can comment

ilario, 2017-06-04:
Related discussion is going on in #137

ilario, 2017-07-04:
smonit was merged in #142, some of the workarounds got implemented in it.

nicopace, 2017-07-05:
@ilario are you sure all the workoarounds already in smonit? @G10h4ck @p4u @altergui can you confirm?

ilario, 2017-07-05:
@nicopace no I'm not

nicopace, 2017-07-05:
Then let's the guys close the issue :)

closed 2017-09-14 by altergui