libremesh/lime-packages #99: Integate tinc into LibreMesh

nicopace, 2017-03-31:
Implement integration with Tinc VPN: http://tinc-vpn.org/

ilario, 2017-04-23:
This sounds like the [GSoC idea](https://wiki.freifunk.net/Ideas#LibreMesh_WPA2_and_802.11s_mesh_encryption_support) of implementing LibreNet6 in LiMe.
 +1

dangowrt, 2017-06-14:
tinc can be great for connecting otherwise unconnected mesh segments, mainly in order to allow management and monitoring. I wouldn't route user traffic for tinc, the performance on those small MIPS CPUs is way to bad. If it's only about having a gateway VPN, I'd rather use tunneldigger which offers near-to wirespeed performance. See also https://github.com/openwrt/packages/pull/4420

lucapost, 2017-08-09:
any update about this? I need integrated vpn for private lan and share private services...

nicopace, 2017-08-12:
No update.
There is a way to deal with this using OpenVPN, but you need to do it manually.
Also, the approach here was not to route internet through it, but actually to allow someone from outside to get into the network... so it is not the case.

p4u, 2017-08-12:
> No update.
> There is a way to deal with this using OpenVPN, but you need to do it
> manually.
> Also, the approach here was not to route internet through it, but
> actually to allow someone from outside to get into the network... so
> it is not the case.
I would say it is the same approach. So the private vpn can be used for
accesing to the mesh network or to encrypt the outgoing traffic of the
mesh to the Internet.

ilario, 2017-11-08:
added Wireguard to the issue title, which should be very efficient

ilario, 2017-11-08:
other options: SoftEther or Fastd

ilario, 2017-11-08:
another option: l2tp

aparcar, 2018-01-23:
I'd vote for wireguard. Now that lime uses 17.01.**4** it's supported by the kernel and should be useable.

ilario, 2018-02-10:
@nicoechaniz suggests to consider [LibreNet6](http://docs.altermundi.net/LibreNet6) integration in LibreMesh as a priority

ilario, 2018-02-10:
A problem of Wireguard is that is [not yet considered stable](https://www.wireguard.com/#about-the-project) but it's receiving a lot of interest, so it could reach stability quite soon. Moreover it's already packaged for LEDE 17.01.4 (see base/wireguard, base/wireguard-tools, luci/luci-app-wireguard, luci/luci-proto-wireguard).

G10h4ck, 2018-02-10:
AFAIK wireguard tunnels are L3 only aka can't be used for routing protocols that needs L2 to send hello messages

ilario, 2018-02-10:
Regarding the size of various packages and their dependencies (measured compiling the armvirt image and comparing with plain lede-17.01.4-lime-default-armvirt-root.squashfs.gz image dimension):
plain: 2733 kB
tinc: +601 kB
babeld: +38 kB
wireguard: +79 kB
wireguard, wireguard-tools, luci-app-wireguard, luci-proto-wireguard: +81 kB

pedro-nonfree, 2018-02-11:
+1 for wireguard. In addition what people said: is extremely easy to configure and its lines are very clear (I configured years ago a tinc site-to-site, I was not very happy what configuration looks like)

Some people did tests to encapsulate L2 traffic inside wireguard

https://lists.zx2c4.com/pipermail/wireguard/2017-March/001156.html

a partner did tests with wireguard + olsr for tunnel broker. He chose olsr as a simple/easy mesh protocol. I would like to know why babel.

ilario, 2018-02-11:
Regarding Babeld, there's an implementation draft from @altergui [here](https://github.com/libremesh/lime-packages/commit/42e65ab1744c72be8c198be7f390a7785614b054)

tekoholic, 2018-03-08:
+1 for wireguard. It is extremely resource-efficient, tinc is (as already mentioned) difficult to configure for users, etc. Attempting, however, to establish a wireguard link to outside provider does not seem to route correctly, in LiMe 17.06 built on chef server. Documentation is scarce, and that which does exist does not correct the routing issues.

ciaby, 2018-05-09:
I'm also interested in having Wireguard as a possible VPN backbone protocol when site-to-site wifi links are not feasible. I already tried to set it up manually, but there is no traffic going to the VPN even with a static route. What should be done to integrate it into LiMe?

ilario, 2018-12-08:
@aparcar I saw [this Tinc package](https://github.com/libremesh/lime-packages/tree/master/packages/tinc)... should we consider this fixed?

ilario, 2020-01-10:
Check out  #579

closed 2020-06-30 by ilario