libremesh/lime-packages #115: Private VPN between nodes over libremesh

lucapost, 2017-04-09:
Hi all, 
    in past time I had private vpn with tincd/fastd running over my mesh nodes, on firmware openwrt based.
Over all nodes i have a bridge for private network with tunnel interface, one ethernet and one wifi interface.
The goal is setup private service using only single device with libremesh and private networks.
Recently i see interesting iusse about: https://github.com/libremesh/lime-packages/issues/92
This is my little features request :-)

Thanks,
Luca.

lucapost, 2017-04-09:
There is https://github.com/libremesh/lime-packages/issues/99 also...

p4u, 2017-04-12:
Thanks, we will have your request into account.

ilario, 2017-04-23:
Encryption and  a private network (now open to external nodes willing to join) can be achieved with AP-Sta WPA2 blackbone (already implemented in LiMe I think), and also with the encryption options of 802.11s (how does it work? with a shared secret passphrase?).

Having a VPN with Tinc was one of the [ideas proposed for GSoC this year](https://wiki.freifunk.net/Ideas#LibreMesh_WPA2_and_802.11s_mesh_encryption_support) and would allow for external nodes to enter the VPN once their key is somehow accepted.

ilario, 2017-05-25:
I would close this issue as duplicate of #99, anyone disagrees?

closed 2017-08-01 by ilario