libremesh/lime-packages #116: LibreMesh release 2017 firmware for 4MB devices

p4u, 2017-04-12:
Using the lime-sdk tool it should be possible to cook binaries for 4MB devices.

There should be two different flavors:

- Minimalist system including WebUI (LuCI)
- Minimalist system without WebUI (with more space to install user stuff)

ilario, 2017-04-23:
Would it be interesting to have a flavor without BMX6 and with LuCI?

p4u, 2017-04-25:
bmx6 is the one who choose the gateway to Internet and other networks in
a standard libremesh setup. So, right now it does not have much sense
IMO. Some structural changes would be needed.

In any case, bmx6 does not make a big difference in the total amount of
flash space for a router.

In the other side, remove LuCi makes sense IMO, there is actually a SDK
profile which does not include it.

On 23/04/17 17:19, Ilario Gelmetti wrote:
> Would it be interesting to have a flavor without BMX6 and with LuCI?
> 

-- 
./p4u

p4u, 2017-04-26:
Achieved with this set of commits: https://github.com/libremesh/lime-packages/commits/feature/minimize_space

And also some lime-sdk modifications: https://github.com/libremesh/lime-sdk/commit/d8cbf7b50c2c540e8c00639cd3abe5e54ae25638

closed 2017-04-26 by p4u