libremesh/lime-packages #117: lime-config do not configure properly ramips network

p4u, 2017-04-13:
As reported in the mailing list there is a problem with the lime-config network interface detection when using it in ramips architecture. 

> root@LiMe-16da88:~# lime-config
> Clearing wireless config...
> Clearing network config...
> Disabling odhcpd
> Cleaning dnsmasq
> Disabling 6relayd...
> /usr/lib/lua/lime/network.lua:155: attempt to index local 'dev' (a nil value)
> stack traceback:
>         /usr/bin/lime-config:45: in function </usr/bin/lime-config:45>
>         [C]: in function 'foreach'
>         /usr/lib/lua/lime/network.lua:192: in function 'scandevices'
>         /usr/lib/lua/lime/network.lua:217: in function
> </usr/lib/lua/lime/network.lua:208>
>         [C]: in function 'xpcall'
>         /usr/bin/lime-config:45: in function 'main'
>         /usr/bin/lime-config:65: in main chunk
>         [C]: ?
> Configuring system...
> Let uhttpd listen on IPv4/IPv6

> root@LEDE:~# ip link
> 1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN qlen 1
>     link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
> 2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state
> UNKNOWN qlen 1000
>     link/ether 78:a3:51:16:da:88 brd ff:ff:ff:ff:ff:ff
> 3: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN qlen 1000
>     link/ether 78:a3:51:16:da:89 brd ff:ff:ff:ff:ff:ff
> 4: wlan1: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN qlen 1000
>     link/ether 78:a3:51:16:da:88 brd ff:ff:ff:ff:ff:ff
> 5: br-lan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state
> UP qlen 1000
>     link/ether 78:a3:51:16:da:88 brd ff:ff:ff:ff:ff:ff
> 6: eth0.1@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue
> master br-lan state UP qlen 1000
>     link/ether 78:a3:51:16:da:88 brd ff:ff:ff:ff:ff:ff
> 7: eth0.2@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue
> state UP qlen 1000
>     link/ether 78:a3:51:16:da:89 brd ff:ff:ff:ff:ff:ff
> 

> root@LEDE:~# iw list| grep Wiphy
> Wiphy phy1
> Wiphy phy0

p4u, 2017-04-13:
The problem is probably here:

https://github.com/libremesh/lime-packages/blob/develop/packages/lime-system/files/usr/lib/lua/lime/network.lua#L196

ilario, 2017-04-18:
A temporary workaround consists in commenting out this line in network.lua:
https://github.com/libremesh/lime-packages/blob/7cda1ad852055e85a3fec73764826ec00ed735bd/packages/lime-system/files/usr/lib/lua/lime/network.lua#L178
which had been introduced two years ago by @G10h4ck in caad3ac.
I copy from the [lime-users mailing list](https://lists.libremesh.org/mailman/listinfo/lime-users) my explanation of the bad behaviour:
> On 04/14/2017 05:53 AM, Juergen Kimmel wrote:
>> Here is
>> /etc/config/network
>> 
>> config device 'lan_dev'
>> option name 'eth0.1'
>> option macaddr '78:a3:51:16:da:88'
>> 
>> config device 'wan_dev'
>> option name 'eth0.2'
>> option macaddr '78:a3:51:16:da:89'
> 
> Ok, now can you try to comment out (prepending "--!") line 178 in
> /usr/lib/lua/lime/network.lua ?
> 
> function owrt_device_parser(section)
> 	dev_parser(section["name"])
> --!	dev_parser(section["ifname"])
> end
> 
> https://github.com/libremesh/lime-packages/blob/develop/packages/lime-system/files/usr/lib/lua/lime/network.lua#L178
> 
> According to LEDE and OpenWrt wikis [1,2], sections of type device in
> /etc/config/network should have an ifname option.
> LibreMesh scripts assume that there is such option and the crash you
> experienced could be due to the lack of an "option ifname 'eth0'" in
> device sections in /etc/config/network on ramips.
> I hope this works... :)
> Ilario
> 
> [1] https://wiki.openwrt.org/doc/uci/network#devices
> [2] https://lede-project.org/docs/user-guide/switch_configuration#devices

dangowrt, 2017-04-18:
We should switch to use our own version of config-generate and use
/etc/board.d stuff instead asap...

p4u, 2017-04-18:
@ilario I thin kyour workaround would break other platforms. 

IMO the workaround should add a first line in dev_parser() function like:
`if dev == nil then return end`

That should solve the crash too.

However I don't really understand what is the point of these sections:

> config device 'lan_dev'
> option name 'eth0.1'
> option macaddr '78:a3:51:16:da:88'

I never saw them before, it's a ramips thing? What happens if eth0.1 is defined here but also in "br-lan" ?

p4u, 2017-04-18:
> We should switch to use our own version of config-generate and use /etc/board.d stuff instead asap...
@dangowrt that is probably right, but it will be a work maybe for the next release

dangowrt, 2017-04-18:
On Tue, Apr 18, 2017 at 03:48:55AM -0700, Pau wrote:
> @ilario I thin kyour workaround would break other platforms.
>
> IMO the workaround should add a first line in dev_parser() function like:
> `if dev == nil then return end`
>
> That should solve the crash [...]

It would have the same mac address as the WAN interface.
Please have a look here to understand how config is generated and
realize that there is no point to re-process the generated config
to generate yet another config.

https://git.lede-project.org/?p=source.git;a=blob;f=package/base-files/files/bin/config_generate;h=6230b0892f04c9853706a06a9547c7e5cc249625;hb=HEAD#l83

commit 81ddc3a9 (2017-04-25): Fix #117
closed 2017-04-25 by p4u