libremesh/lime-packages #119: Dependency problem with luci-lib-jquery-1-4 and luci-lib-dracula

ilario, 2017-04-22:
`luci-lib-jquery-1-4` and `luci-lib-dracula` are dependencies of `luci-app-batman-adv` and of `luci-mod-status` (both are dependencies of lime-webui). 
They were removed from lime-packages in 406f843 with the reason that "[they are] Now available on oprnwrt-routing".
In the [mentioned repository](https://git.lede-project.org/?p=feed/routing.git;a=summary) there's no such package satisfying these dependencies. The required files are indeed included in a package named `luci-app-bmx6`.
The solution can't be to insert a dependency for `luci-app-bmx6` as it would pull in the whole bmx6.

p4u, 2017-04-24:
They are defined inside luci-app-bmx6 Makefile but they are different packages and can be selected independently to luci-app-bmx6

ilario, 2017-04-24:
Ah!!!?
Ok.

ilario, 2017-04-25:
I still get errors using lime-build, even after fixing feeds for lime-build (see [this PR](https://github.com/libremesh/lime-build/pull/22)):
```
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for lime-basic:
 *      luci-lib-jquery-1-4 *   luci-lib-dracula *      luci-lib-jquery-1-4 *
 * opkg_install_cmd: Cannot install package lime-basic.
make[3]: *** [package/install] Error 255
make[2]: *** [/home/igelmetti/lime-build/build/src/staging_dir/target-mips_24kc_musl-1.1.16/stamp/.package_install] Error 2
```
Why?

p4u, 2017-04-26:
Which openwrt-routing branch is your lime-build clone using?

It should be "master" (not any tag or branch).

On 25/04/17 19:33, Ilario Gelmetti wrote:
> Reopened #119.
> 

-- 
./p4u

ilario, 2017-04-26:
Ah! 
Ok, thanks, that was the problem.

closed 2017-04-26 by ilario