libremesh/lime-packages #123: ieee80211s does not work if AP enabled

p4u, 2017-05-05:
This /etc/config/lime works

> config lime network
> 	list protocols ieee80211s
> 	list protocols lan
> 	list protocols anygw
> 	list protocols batadv:%N1
> 	list protocols bmx6:13
> 
> config lime wifi
> 	list modes ieee80211s
> 

But if ap and/or apname are enabled, the mesh interface is not enabled properly by netifd 

> config lime wifi
> 	list modes ieee80211s
> 	list modes ap
> 	list modes apname

dangowrt, 2017-05-08:
see also [LEDE FS#619](https://bugs.lede-project.org/index.php?do=details&task_id=619).
fixed by neoraider in his staging tree by [commit 6db7dcd74a](https://git.lede-project.org/?p=lede/neoraider/staging.git;a=commit;h=6db7dcd74aaf23865ff19b515daf8c534414adaf)

p4u, 2017-05-11:
Temporary fixed on lime-sdk until 17.01.2 is released

https://github.com/libremesh/lime-sdk/commit/7971d28d40ba456b73214bb482f8d97d156ff66e

closed 2017-05-11 by p4u