libremesh/lime-packages #124: Add a way to select ieee80211s on menuconfig

p4u, 2017-05-16:
According to the discussion had on the mailing list, the idea is to migrate the default libremesh link layer from AdHoc to 802.11s. To this end, a way to easy select 11s or adhoc on the firmware would be needed. This is also required to lime-sdk, to be able to produce both versions of the firmware.

G10h4ck, 2017-05-18:
We could do this by splitting lime-*-adhoc and lime-*-80211s out of lime system, so they can be selected/un-selected in menuconfig

ilario, 2017-05-18:
@G10h4ck and enabling both in lime-defaults?

p4u, 2017-05-18:
> We could do this by splitting lime--adhoc and lime--80211s out of lime system, so they can be selected/un-selected in menuconfig
Sure. We can create a special metapackage named "lime-profiles" which contains a set of packages, for instance:

1. lime-11s
2. lime-adhoc

These packages will have a (for instance) uci-defaults script to set 11s or adhoc properly on the first boot.

Or another option is (what I already did) to use the network-profiles git repository:
https://github.com/libremesh/network-profiles/commit/f0992296ed7248cc76cb102a5dc26a2fd7c0e247

So right know a libremesh 11s firmware can be cooked with lime-sdk:
`./cooker -c ar71xx/generic --profile=tl-wdr3600-v1 --flavor=lime_zero --community=libremesh/11s`

p4u, 2017-05-18:
> @G10h4ck and enabling both in lime-defaults?

I don't think this is a good idea...

ilario, 2017-05-18:
Actually this is the way we adopted for routing protocols selection: all of them are selected in lime-defaults (except BGP) but not included in lime-basic Makefile. So +1 for me.

@p4u I would call it ieee80211s instead of 11s, just for keeping coherence with [current prono name](/libremesh/lime-packages/tree/develop/packages/lime-system/files/usr/lib/lua/lime/proto)

p4u, 2017-05-18:
> Actually this is the way we adopted for routing protocols selection: all of them are selected in lime-defaults (except BGP) but not included in lime-basic Makefile. So +1 for me.

But we are not using 11s as a routing protocol, neither "adhoc" is a routing protocol. If adhoc and ieee80211s protocols are enabled at same time, network meshing won't work probably.

> @p4u I would call it ieee80211s instead of 11s, just for keeping coherence with current proto name. Alternatively let's rename the proto name to 11s.

I prefer to use 11s for the network-profile repository since it is what people better understand.

G10h4ck, 2017-05-18:
> I prefer to use 11s for the network-profile repository since it is what people better understand.

I believe this is a wrong assumption, if you look for 11s on any search engine you will see that none point to what we refer to 11s...

I vouch for ieee80211s

p4u, 2017-05-25:
> I believe this is a wrong assumption, if you look for 11s on any search engine you will see that none point to what we refer to 11s...
> 
> I vouch for ieee80211s

Same as if you look for "adhoc" "ap" or "apname". However if you include the word "wifi" you will find them, also 11s.

But I don't really care, so ieee80211s is fine for me.

p4u, 2017-05-25:
I remove the release-blocker flaw, as it is partially solved by using network-profiles in cooker.

ilario, 2018-02-04:
Do we still want this or can we close the issue?

closed 2018-12-08 by ilario