libremesh/lime-packages #127: Documentation needed for WiFi client mode

ilario, 2017-05-18:
Currently [wifi client mode](/libremesh/lime-packages/blob/develop/packages/lime-system/files/usr/lib/lua/lime/mode/client.lua) allows you to configure in `/etc/config/lime` both WAN or LAN network (for discussion see issue #47 and description of commit 6ded289).
We need to add in `/etc/config/lime-example` (the location for basic documentation) a reminder and an example for this mandatory LAN/WAN setting.

ilario, 2017-07-23:
@p4u @G10h4ck how can I predict the linux_name of the client interface?
This is needed for writing an example of documentation.
For making it clear, an example of minimal client configuration could be:

```
config wifi wificlientradio
	list modes 'client'
	option channel_2ghz 'auto'
	option client_ssid 'SomeWiFiNetwork'
	option client_key 'SomeWPApskPassword'
	option client_encryption 'psk'

config net wificlientnet
	option linux_name '??????????????????'
	list protocols 'wan'
```

what to use for linux_name?

G10h4ck, 2017-07-26:
> what to use for linux_name?
wlanX-client remember to substitute X with the number of the interface

in general the names are composed this way

wlan${RADIO_NUMBER}-${LIME_WIFI_MODE_NAME}

ilario, 2017-07-27:
So it would be safe to use `wlan0-client` in lime-example and explain in comments that the user have to manually verify this via ssh or web interface?

ilario, 2017-08-21:
Correct me if I'm wrong:

* when using WiFi client mode to connect to a **non-LiMe** AP, `list protocols 'wan'` should be used;
* when connecting to a **LiMe AP** (AP-sta could be a good idea for node to node connections, mainly when using hardware which doesn't suppot ieee80211s or adhoc), `list protocols 'wan'` should be **avoided** as a NATting would create loops in the network, instead something like `list protocols lan
	list protocols anygw
	list protocols batadv:%N1
	list protocols bmx6:13` should be used.

closed 2017-09-15 by p4u

ilario, 2017-09-18:
Can we discuss why the assumptions in my last comment are wrong?
I suppose they are 'cause 12ec6b5 deleted the wan/lan distinction, but this was not explained anywhere.
@G10h4ck pls

G10h4ck, 2017-09-18:
You use wan to get upstream connectivity, not to connect to a libremesh network, while using lan would require bridging and wifi client interface are not capable of that in a standard way

ilario, 2017-09-18:
> You use wan to get upstream connectivity, not to connect to a libremesh network,

Sure, no doubt on this.

> while using lan would require bridging and wifi client interface are not capable of that in a standard way

Sorry for the ignorance, but why?

The use-case of this is a network either made of devices which cannot do adhoc or ieee802.11s, or a network with an hierarchic topology (few-supernodes-backbone connected to many-leafnodes-clients, only AP-sta connections like most of guifi.net and ninux.org communities) where you use no mesh connections, just AP-sta also for backbone.

G10h4ck, 2017-09-18:
https://wiki.openwrt.org/doc/howto/clientmode#bridged_client_mode_issues

ilario, 2017-09-23:
> In that case you don't run lan protocol on the AP-STA links and interfaces but other protocols such as bmx or batman-adv

Ok :D 
So in that case the configuration would be something like:

```
config net wirelessclient
	option linux_name 'wlan0-client'
	list protocols ieee80211s
	list protocols anygw
	list protocols batadv:%N1
	list protocols bmx6:13
```

without lan, right?

G10h4ck, 2017-09-24:
On Saturday, 23 September 2017 20:51:31 CEST Ilario Gelmetti wrote:
> list protocols ieee80211s

Exept for this it probably do nothing at runtime, but can create confusion in 
the documentation

> without lan, right?

yeah

commit 12ec6b5c (2017-09-15): Wireless client usage example

Fix #127
Obsolete PR 223
commit 8d965f82 (2017-09-15): Merge pull request #232 from libremesh/hotfix/wifi_client_doc-127

Add example usage for wifi client mode. Fix #127
commit 9ecde307 (2017-09-15): Add documentation for wifi client mode in lime-example, fix #127