libremesh/lime-packages #132: Make Name Resolution work

nicopace, 2017-05-21:
note in spanish:
@altergui: resolucion de nombres (o alguna forma de conectarse con nodos) para los humanos q estan dentro de la red

ilario, 2017-05-25:
How should this work?
I thought we already had this (but I never used it).

ilario, 2017-05-25:
Is this related to #118?

nicopace, 2017-05-29:
no.
this is related to hostname and domain name resolution for both your cloud and other clouds.
like, if your node's hostname is 'nico' and network domain is 'quintana.libre.social', then everyone should be able to access it as nico.quintana.libre.social

FreifunkUFO, 2017-07-03:
its not a bug of libremesh firmware, but a possibile enhancement of your network setup.

so (seen from the internet), you network domain has to be 

* able to resolve your personal name (nico.quintana.libre.social) to an ipv4 OR/AND ipv6
* libremesh gateway node has to route that traffic
* libremesh gateway node has to firewall that traffic (do an accept)

do you have only one ipv4-ip on the gateway? so that would be difficult. you will need an http.proxy on the gateway node to direct the http-traffic to your special node/client. and its only working for http

do you have an small ipv6-subnet on the gateway? so you may think ipv6 sucks, arent there some more ips or do we reorder/NAT them?

do you have an big ipv6-subnet on the gateway? so it sounds nice. you still have to take a look at your firewall rules, even when your DNS with this ipv6-adresses is working properly..

ilario, 2018-12-08:
related: #193 #303 #390

ilario, 2019-10-04:
From *inside* of the network it already works.
What @FreifunkUFO writes is related to the possibility to access the host from outside of the network (internet) and is the object of #133.
Closing this.

closed 2019-10-04 by ilario