libremesh/lime-packages #134: Howto centralized log

nicopace, 2017-05-21:
Notes:

howto"  log centralizado y/o collectd, en un router con pendrive (o  rPi). ya  hay presettings de chef (comentados) para el log central. Esto también  habría que coordinar con Conraad para que el LibreServer tenga una  interfaz web piola para ver data de Collectd.

ilario, 2017-05-25:
So what's missing here, documentation?

nicopace, 2017-05-29:
i guess this is more for Conraad (LibreServer), but like a reminder.
@nicoechaniz can explain further

ilario, 2017-08-01:
If this is not regarding LibreMesh I would close the issue...

nicopace, 2017-08-01:
@nicoechaniz ?
Also, It's something related to LibreMesh, and documentation on how to setup a monitoring server on this context.

pedro-nonfree, 2017-09-24:
In the past, I succesfully redirected logs of some qMp antennas to a central server. This is very useful when you cannot access antenna to see what's happenning, but you want to see the latest logs before crash. But now I would like to do again and I don't remember. Are you interested in this feature? As far as I remember, this is a redirection of the log to a rsyslog server. After that, I think you can process the log as you want (collectd, etc.)

The references right now:

- https://wiki.openwrt.org/doc/howto/log.essentials
- https://wiki.openwrt.org/doc/howto/log.overview

ilario, 2018-12-08:
Updates on this "reminder"? Can we close?

ilario, 2019-10-04:
The collection of logs is part of https://github.com/libremesh/lime-log-review but we still have to discuss on how to collect these logs.
Does it make sense to do it with a Prometheus exporter?

ilario, 2020-06-05:
I just noticed that some communities already use collectd: 
https://github.com/libremesh/network-profiles/blob/master/quintanalibre.org.ar/generic/files/etc/collectd.conf

ilario, 2023-03-03:
Closing as utterly inactive.

closed 2023-03-03 by ilario