libremesh/lime-packages #135: Adapt lime-config to LEDE json

nicopace, 2017-05-21:


ilario, 2017-05-21:
Eh? LEDE json?

nicopace, 2017-05-21:
Mmm... sorry, I have no information on this... I just transcripted it from the LibreStack Pad we had... Maybe @altergui or @p4u  can help to understand the meaning of the issue

dangowrt, 2017-06-14:
Instead of re-writing the configuration **based** on the configuration itself (which works only once, for obvious reasons), config generation should instead be based on `/etc/board.json`. See https://github.com/lede-project/source/blob/e2fd98793e60306a06427a9374e893d380373ef1/package/base-files/files/bin/config_generate for how config generation currently works.
The goal is to make lime-config more easy to debug, make it possible to re-run it and hence unlock keeping configuration during sysupgrade (such as PPPoE or WiFi client-mode credentials which should be not part of a profile somewhere in a repository).

G10h4ck, 2019-02-21:
If I remember well @aparcar did this for the most convenient parts, like taking lan/wan interface name

closed 2019-02-21 by G10h4ck