libremesh/lime-packages #144: Ath9k radio, MCS stock workaround

p4u, 2017-06-07:
According to the information provided by @nicoechaniz (https://github.com/libremesh/network-profiles/blob/master/quintanalibre.org.ar/comun/etc/uci-defaults/92-cron-workaround-tasks), some Atheros radios (the ones from Tplink 3500 and Tplink 842) stock in a strange state where the driver thinks MCS can be higger than 7 but it might be not. In consecuence throughput is slowly reducing until it almost lost connectivity with other nodes. 

Performing a scan (in more than one frequency) fixes this issue and the radio goes to the normal state again.

It would be needed to find a way to identify this problem (when is it happening) and perform an acction to fix it and save some debug information to fix it upstream. Smonit might do this job (#137).

Current workaround is a cron executed every 5 minutes:

```
iw wlan1-adhoc scan freq 5745 5825
iw wlan0-adhoc scan freq 2412 2457iw wlan1-adhoc scan freq 5745 5825
iw wlan0-adhoc scan freq 2412 2457
```

nicoechaniz, 2017-06-10:
this is a very tricky bug.

Here in QuintanaLibre we are able to tell that the MCS rate negotiated is unrealistic because of the split MIMO chains scenario. A node with its chains splitted in opposite directions should never negotiate an MCS15 rate with any distant node, but we have found this to happen regularly.

A node acting properly will show something like this when doing an iw station dump:

```
root@oncelotes:~# iw wlan1-adhoc station dump
Station 64:70:02:3d:84:03 (on wlan1-adhoc)
	inactive time:	10 ms
	rx bytes:	1297596979
	rx packets:	6589765
	tx bytes:	2513743403
	tx packets:	2049296
	tx retries:	449374
	tx failed:	18
	rx drop misc:	12870
	signal:  	-64 [-64, -85] dBm
	signal avg:	-67 [-67, -87] dBm
	tx bitrate:	135.0 MBit/s MCS 6 40MHz short GI
	rx bitrate:	108.0 MBit/s MCS 5 40MHz
	expected throughput:	40.191Mbps
```

It's visible there that the chains are split (signal level is very different on each chain) and the negotiated MCS in this scenario can not go beyond MCS7.

So, a not so tidy way to monitor this effectively would require parsing iw station dump output and reacting to not invalid MCS rates.

dangowrt, 2017-06-29:
@ecsv @nbd168 This ticket does provide some interesting hints towards where to look for the bug...
I recon it is related to #130 .
Can someone confirm that this issue (and in case it's a different issue, as well as #130) is affecting Ad-Hoc mode only, ie. did anyone see this occuring when using 11s?

FreifunkUFO, 2017-07-03:
i will try to take a look on it

root@211-54:~# iw dev wlan0 station dump | grep sign
        signal:         -88 [-92, -91] dBm
        signal avg:     -88 [-90, -91] dBm
        signal:         -56 [-56, -66] dBm
        signal avg:     -56 [-56, -66] dBm
        signal:         -83 [-84, -91] dBm
        signal avg:     -83 [-83, -91] dBm
        signal:         -89 [-91, -93] dBm
        signal avg:     -88 [-90, -93] dBm
        signal:         -68 [-73, -69] dBm
        signal avg:     -67 [-73, -68] dBm
        signal:         -69 [-70, -79] dBm
        signal avg:     -68 [-68, -78] dBm
        signal:         -70 [-72, -74] dBm
        signal avg:     -69 [-72, -71] dBm
        signal:         -89 [-93, -91] dBm
        signal avg:     -88 [-92, -91] dBm

nicoechaniz, 2017-07-04:
I still haven't had the time to flash a portion of our net to use 802.11s and test this. Will report as soon as we have data.

ilario, 2017-08-01:
Are there any news on this?
Is the new [frozen repo](/libremesh/compat-wireless) going to fix this?

p4u, 2017-08-01:
https://github.com/libremesh/compat-wireless

ilario, 2017-08-01:
Thanks Pau, I already put that link in my previous comment. Does this mean we can close this?

p4u, 2017-08-01:
Right, sorry.

I'm not sure if this will fix the issue, let's wait until it is tested.

ilario, 2017-08-03:
@p4u how can we use that repository with lime-sdk?

ilario, 2017-09-25:
We didn't use that repository for the 17.06 release, right?
Means that this problem was not actually happening?
Shall we close this?

FreifunkUFO, 2017-10-24:
so how are your tests? and what is the special about that "frozen repo"? it was copied, so i cant see the differencies to the "normal" lede-stable or -trunk.

@dangowrt i found some freifunk-gluonners having "this" (?) bug on 11s too :-( 
https://github.com/freifunk-gluon/gluon/issues/1148

p.s. when i had similiar problem recently (lede-trunk from february), iinstead of a wifi-scan the node was able to go back to work.. looking at ANI: i have tried turning it off and on again, successfully.

echo 0 >/sys/kernel/debug/ieee80211/phy0/ath9k/ani

echo 1 >/sys/kernel/debug/ieee80211/phy0/ath9k/ani

please could anyone else test this instead of wifi-scans!?

p4u, 2017-11-01:
We abandoned the idea of frozen compat-wireless repository because it did not make a difference. So that repository is "obsoleted", please do not use it.

About what @FreifunkUFO proposes, the only one who can reproduce this error is @nicoechaniz right now.

ilario, 2018-02-04:
> We abandoned the idea of frozen compat-wireless repository because it did not make a difference. So that repository is "obsoleted", please do not use it.

@p4u so can I delete that repository?

The solution of this problem is being reworked by @nicopace in packages/cotonete merged in #305

dangowrt, 2018-04-25:
Just confirming that this bug still exists, even when using 802.11s and also when using wpa_supplicant for SAE and/or DFS. Seems like we need some kind of interface watchdog... cotonete doesn't work for me and seems to be outdated. I also noted that in case of a deaf phy in 802.11s SAE mode the peers disappear entirely from the assoclist, so the logic in cotonete wouldn't work in that case even if the script did work...
Maybe we should write a general purpose interface watchdog which should:
* monitor interface state and counters (`ubus call network.device status`)
* keep track of associated stations on wireless interfaces (`ubus call iwinfo assoclist '{ "device": "wlan0-mesh" }'`
* detect rx-stuck based on counters and assumed interface state
* trigger fixup routine based on interface type (ie. send ubus events to netifd)

This would not be hard to do using libubox and ubus in C, but still quite some work by the end of the day. But it would be a valuable add-on for netifd for the time being. The very same ubus rpc interfaces could also be used to do the same in Lua, probably in less time, but at the cost of reusability.
This whole Atheros platform ends up being very disappointing because it sucks at the two most important aspects: ethernet phys still gets stuck randomly (on ubnt hardware) and ath9k wifi rx just looses it entirely sometimes...

ilario, 2019-04-17:
Is this still present?
@dangowrt @G10h4ck @p4u @FreifunkUFO @nicoechaniz @nicopace

spiccinini, 2019-04-24:
As far as I know this issue is still present at least in 18.06.1 and LiMe does not provides a working workarround as cotonete is reported to be outdated/not working.

nicopace, 2019-04-24:
cotonete definitely needs some help.
the strategy for detecting the issue was accurate and fast, but no additional coding was done to take actions, nor to further debug a bug that was detected by some that made cotonete eat the cpu.

aparcar, 2019-04-24:
@dangowrt

ilario, 2019-10-04:
Cotonete was removed from lime-packages with #584, its new house (if anyone wanted to contribute) is: https://github.com/nicopace/cotonete

> a bug that was detected by some that made cotonete eat the cpu

@nicopace do you have a more detailed report on this?

nicopace, 2019-10-08:
It was a POC that we did with @nicoechaniz to explore how to work around the ath9k_bug (or any other bug that could arise with radios).
Basically, cotonete sends a ping over link local to one each neighbour. if the ping shows changes in the packages transfered, then the radio is working. if not, it should try a series of workarounds.
We did some tests (a while ago), and it was working... but no further work was done on it yet.

pedro-nonfree, 2020-06-29:
Another workarounds to inspire:

- following the strategy of *holding the good things plus improving others*
  - extend the support for the current release, specially for this targeted devices. Maybe increasing versions of the rest, specially the routing protocol. Or using the same routing protocol version as the old release.
  - using old version of ath9k in the new release, maybe only for the targeted devices
- move this bug from release-blocker to well-known bug, and keep on working of a work in progress solution
  - also include information, not only about "what should happen" but also about throughput "before and after" the problem.

dangowrt, 2020-06-29:
On 19.07.3 things look extremely stable until now.
So this is Atheros AR9340 Rev:2 of a NanoStation M XW which previously got stuck almost daily running LEDE 17.01.
We go the work-around to do a wifi scan every now and then active, didn't have any connection drops in weeks:
```

BusyBox v1.30.1 () built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 19.07.3, r11063-85e04e9f46
 -----------------------------------------------------
  ___   __ __                _______             __
 |   |_|__|  |--.----.-----.|   |   |-----.-----|  |--.
 |     |  |  _  |   _|  -__||       |  -__|__ --|     |
 |_____|__|_____|__| |_____||__|_|__|_____|_____|__|__|

 ------------------------------------------------------
 LiMe master development (master rev. 20888155 20200604_1356)
 ------------------------------------------------------
 http://libremesh.org
 ------------------------------------------------------
root@ww-dach:~# ifconfig wlan0-mesh
wlan0-mesh Link encap:Ethernet  HWaddr 80:2A:A8:06:95:71
          inet6 addr: fe80::822a:a8ff:fe06:9571/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1536  Metric:1
          RX packets:557710458 errors:0 dropped:5 overruns:0 frame:0
          TX packets:968256904 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:80711496881 (75.1 GiB)  TX bytes:1337731787447 (1.2 TiB)

root@ww-dach:~# iw dev wlan0-mesh station dump
Station fc:ec:da:60:6b:ee (on wlan0-mesh)
	inactive time:	50 ms
	rx bytes:	2719016835
	rx packets:	24175863
	tx bytes:	10307143049
	tx packets:	10430212
	tx retries:	964557
	tx failed:	49
	rx drop misc:	41569
	signal:  	-66 [-67, -71] dBm
	signal avg:	-65 [-66, -69] dBm
	Toffset:	18446743260039534759 us
	tx bitrate:	78.0 MBit/s MCS 12
	rx bitrate:	43.3 MBit/s MCS 4 short GI
	rx duration:	0 us
	last ack signal:28 dBm
	expected throughput:	33.507Mbps
	mesh llid:	0
	mesh plid:	0
	mesh plink:	ESTAB
	mesh local PS mode:	ACTIVE
	mesh peer PS mode:	ACTIVE
	mesh non-peer PS mode:	ACTIVE
	authorized:	yes
	authenticated:	yes
	associated:	yes
	preamble:	long
	WMM/WME:	yes
	MFP:	        yes
	TDLS peer:	no
	DTIM period:	2
	beacon interval:100
	connected time:924396 seconds
Station 24:a4:3c:44:c7:ff (on wlan0-mesh)
	inactive time:	10 ms
	rx bytes:	27816073592
	rx packets:	159454677
	tx bytes:	363492947295
	tx packets:	256035112
	tx retries:	21417903
	tx failed:	4
	rx drop misc:	170731
	signal:  	-59 [-61, -64] dBm
	signal avg:	-57 [-59, -62] dBm
	Toffset:	18446742779125224291 us
	tx bitrate:	104.0 MBit/s MCS 13
	rx bitrate:	117.0 MBit/s MCS 14
	rx duration:	0 us
	last ack signal:41 dBm
	expected throughput:	37.994Mbps
	mesh llid:	0
	mesh plid:	0
	mesh plink:	ESTAB
	mesh local PS mode:	ACTIVE
	mesh peer PS mode:	ACTIVE
	mesh non-peer PS mode:	ACTIVE
	authorized:	yes
	authenticated:	yes
	associated:	yes
	preamble:	long
	WMM/WME:	yes
	MFP:	        yes
	TDLS peer:	no
	DTIM period:	2
	beacon interval:100
	connected time:520253 seconds
root@ww-dach:~#
```

spiccinini, 2020-06-29:
Yes, it is running fairly stable with the scan workarround (we are using the package `wifi-unstuck-wa` for this). 
I am inclined to document it as a known-issue / workarround.

spiccinini, 2020-08-26:
Closing this as we are very stable with provided workarrounds.

closed 2020-08-26 by spiccinini

ilario, 2020-08-27:
Amazing!
So are we going to include the workarounds in releases? Totally fine for me.
Are we going to not use [Cotonete](https://github.com/nicopace/cotonete) (whose development was not active anyway)?

pedro-nonfree, 2020-08-27:
I am curious, that workarounds ([is this, right?](https://github.com/nicopace/cotonete)) are useful in general for the openwrt community? Should be moved to upstream? If this has been extensively tested, should beta be removed? re-added to lime-packages?

ilario, 2020-08-27:
> I am curious, that workarounds ([is this, right?](https://github.com/nicopace/cotonete)) are useful in general for the openwrt community? Should be moved to upstream? If this has been extensively tested, should beta be removed? Should it be re-added to lime-packages (#584)?

Sorry @pedro-nonfree, I realise now that my comment was confusing. What you linked is Cotonete.
Cotonete was a way to **detect** when a radio become deaf and it had still a few bugs to fix before being production ready.

The workaround that @spiccinini refers to is this one:
https://github.com/libremesh/lime-packages/blob/master/packages/wifi-unstuck-wa/files/usr/lib/lua/wifi_unstuck_wa.lua
which is packaged in LibreMesh's `wifi-unstuck-wa` package.

I have no opinion on whether the `wifi-unstuck-wa` package should be included in OpenWrt's repositories. On one side it sounds ridiculous to not fix such an issue and to include a workaround, on the other side i there is no better solution the workaround is surely useful to plenty of people.