libremesh/lime-packages #152: bmx6 VLAN not created on wired Ethernet interfaces

dangowrt, 2017-06-14:
While adding an 802.1ad VLAN for BMX6 underneath the br-lan bridge works on devices with a more complex integrated switch (such as TL-WR1043ND or TL-W9870, ifname being `eth0.1`), the `config device` section for that VLAN isn't created on more simple devices such as TL-WR841N (ifname for lan is `eth0` there).

p4u, 2017-06-14:
After some bad experiences on using VLAN tags on Ethernet with BMX6 unidirectional tunnels and MTU. We decided to remove the 802.1ad VLAN from the Ethernet interfaces. This was implemented on the bmx6-mtu-anticlimax branch which will be merged probably today to develop. On my testbed anticlimax and BMX6 over Ethernet works well.

https://github.com/libremesh/lime-packages/tree/hotfix/bmx6-mtu-anticlimax

ilario, 2017-08-01:
Ok, closing this as it was intended as a feature.

closed 2017-08-01 by ilario