libremesh/lime-packages #157: hostname and ssid length are different

nicopace, 2017-06-14:
when you change the name of the node, it truncks the ssid to a certain length and the hostname to another length.

expected behaviour: to be both the same

actual behaviour: names are different

ilario, 2017-06-19:
Mh? Aren't hostname and ESSID independently defined?

nicopace, 2017-06-19:
actually not. in lime we have a config were it is defined in the same place.
this is actually a very simple bug, but just left it with the 'quickfix' tag for any newcommer to be able to easily join and solve it :)

ilario, 2017-07-14:
do you mean  that `option apname_ssid 'LibreMesh.org/%H'` is truncated to a different length than `option hostname 'LiMe-%M4%M5%M6'` when hostname is customized?
Do you have a clue about where does this happen?

nicopace, 2017-08-01:
I don't konw where that is configured in /etc/config/lime ... i did it through the simple config->advanced->hostname LuCI interface

altergui, 2017-08-02:
> actually not. in lime we have a config were it is defined in the same place.
> this is actually a very simple bug, but just left it with the 'quickfix'
> tag for any newcommer to be able to easily join and solve it :)

maybe it's not a bug?

SSID is restricted to 32 characters length, that's why it's truncated
(else, AP with name doesn't come up)

hostname doesn't have that restriction, and it's used elsewhere (like
bmx6, dns, etc) so i'd leave it untruncated

nicopace, 2017-08-02:
not a bug... maybe with the apname it is complex because it is prefixed with the network name, so it gets shrinked even more... 
I get it ... let's leave it that way.
Thanks for the input.

closed 2017-08-02 by nicopace