libremesh/lime-packages #159: Batman web interface does not draw the graph

nicopace, 2017-06-14:
expected behaviour: graph of nodes at 
http://thisnode.info/cgi-bin/luci/batman/graph

actual behaviour: nothing is shown
[image]

code involved:
https://github.com/libremesh/lime-packages/blob/develop/packages/luci-app-batman-adv/files/usr/lib/lua/luci/controller/batman.lua

FreifunkUFO, 2017-07-03:
i tried that with former (karl helga openwrt cc) libremesh version firmware, there also was no graph.

ilario, 2017-08-04:
If I get this right, it's a duplicate of #110 and #1.
All of these issues depends on the fact that luci-app-batman-adv uses `batctl vd` for obtaining data, but this feature was moved to `batadv-vis` command which is part of the `alfred` package which is installed in all LibreMesh flavors (gets selected by dnsmasq-distributed-hosts and dnsmasq-lease-share packages).
So solving this could be as easy as replacing "batctl vd" for "batadv-vis" [here](/libremesh/lime-packages/blob/develop/packages/luci-app-batman-adv/files/usr/lib/lua/luci/controller/batman.lua#L75) (to be verified).

p4u, 2017-08-13:
PR on this issue would be very welcome :)

ilario, 2017-09-21:
Closed by #225

closed 2017-09-21 by ilario