libremesh/lime-packages #161: firmware upgrade via webui

FreifunkUFO, 2017-06-15:
i have my tplink 850 with new official libremesh firmware from 4 days ago.
when trying firmware upgrade via webui to an even newer firmware i got errors:

/usr/lib/lua/luci/dispatcher.lua:460: Failed to execute call dispatcher target for entry '/lime/flashops/sysupgrade'.
The called action terminated with an exception:
/usr/lib/lua/luci/template.lua:74: Failed to load template 'admin_system/applyreboot'.
Error while parsing template '/usr/lib/lua/luci/view/admin_system/applyreboot.htm':
No such file or directory
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:460: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:141: in function </usr/lib/lua/luci/dispatcher.lua:140>

p4u, 2017-06-29:
Is this error still appearing?

FreifunkUFO, 2017-06-29:
im confused with libremesh domains and subdirectories.. 
which one is the most official libremesh testing firmware this week?

nicopace, 2017-06-29:
repo.libremesh.org/lime-17.04/targets/

p4u, 2017-06-30:
The last release firmware is always here: http://repo.libremesh.org/current/

FreifunkUFO, 2017-07-03:
the firmware-files are 2 weeks old :-/ no changes/bugfixes on libremesh these weeks?

and, there ist more than one file per device/flavour, f.i.

http://repo.libremesh.org/current/targets/ar71xx/generic/tl-wr710n-v1/lime_default/lede-17.01.2-libremesh-ar71xx-generic-tl-wr710n-v1-squashfs-factory.bin
vs.
http://repo.libremesh.org/current/targets/ar71xx/generic/Default/lime_default/lede-17.01.2-libremesh-ar71xx-generic-tl-wr710n-v1-squashfs-factory.bin

FreifunkUFO, 2017-07-03:
ok: firmware upgrade now runs without above issue

closed 2017-07-03 by FreifunkUFO

FreifunkUFO, 2017-07-03:
not that nice webui style, but its running...
[image]